pimoroni/enviro #4: Provisioning passwords cannot contain equals signs

exorsyst, 2022-08-08:
When provisioning a Pico Grow from new or via a reset the WIFI password section will fail if an equals sign is used.  Shell reports the following and the provisioning process halts.

```
Task exception wasn’t retrieved
future: coro= <generator object ‘_handle_request’ at 20015990>
Traceback (most recent call last):
File “uasyncio/core.py”, line 1, in run_until_complete
File “phew/phew/server.py”, line 254, in _handle_request
File “phew/phew/server.py”, line 142, in call_handler
File “enviro/provisioning.py”, line 71, in provision_step_3_logging
File “enviro/helpers.py”, line 182, in get_config
File “enviro/helpers.py”, line 169, in get_values_from_file
File “”, line 1
SyntaxError: invalid syntax
```

There is the potential for other characters to cause the same issue but I could only find issue with equals.

Gadgetoid, 2022-08-08:
Looks like a bug in `phew` where form data is decoded using the same assumptions as a querystring, except the form data will not be urlencoded so it'll split on the `=` and break: https://github.com/pimoroni/phew/blob/29909a52917c621680f262bde0018d5189c6b1c6/phew/server.py#L250

Gadgetoid, 2022-08-08:
If you want to try and hotfix your board you can connect it to a computer and use Thonny to change `phew/phew/server.py` line 34 from:

```python
key, value = parameter.split("=")
```

to:

```python
key, value = parameter.split("=", 1)
```

And see if that helps!

MuddyDogs, 2022-08-09:
This is just an FYI. I've had an issue that may be related to this one.
The UI doesn’t like it if your WiFi Access point has a space in the name, and the same for the password. This is true for both of mine.. It’s recording them in the config.py file as a “+” rather than a “ “, so if it was “My Access Point” it gets recorded as “My+Access+Point”.
I got round it manually editing the values in the config.py file, and it works fine now.

Gadgetoid, 2022-08-09:
Any ideas on the above @lowfatcode - this would strike me as an instance of it *being* URL encoded, rather than not?

exorsyst, 2022-08-09:
> ```python
> key, value = parameter.split("=", 1)
> ```

This gives me the same error as in the first example

lowfatcode, 2022-08-09:
Whoops, this is caused by a rather dumb bug here: https://github.com/pimoroni/phew/blob/29909a52917c621680f262bde0018d5189c6b1c6/phew/server.py#L16

Note that the result of the `replace()` call is not written back into `text` so the line has no effect at all.

I'll fix that up and commit the change right away.

closed 2022-08-09 by lowfatcode