pimoroni/enviro #6: Ability to synchronise readings from multiple Enviro modules

lowfatcode, 2022-08-10:
Currently scheduling is always relative (e.g. "every five minutes"). This makes it tricky to setup multiple Enviro boards that collect data at the same time.

Options:

- should "every five minutes" actually be locked to clock time? I.e. at 5 minutes past, 10 minutes past, etc? This would be a simple change that doesn't add new configuration options and logically seems sensible too.
- should the user be able to specify fixed times when readings occur? So if they say "hourly" should it be possible to specify that it's always taken at 15 minutes past the hour?

We definitely want to avoid making the configuration overly complex with too many options. It's harder for the user to reason about.

ExperiMentor, 2022-08-17:
Seems worthwhile implementing, even if simply (without complete versatility).

Given that people can choose unusual sampling frequencies (eg, every 23 minutes), it could simply be implemented as 'make the first measurement at an integer multiple of the interval past the hour' - at 00, 23 or 46 in this example - instead of always "now" as at present. That would satisfy most needs

lowfatcode, 2022-09-02:
Readings are now synchronised to clock time rather than just relative to last boot up. 

This means, for example, if you pick readings every 20 minutes you'll get one at roughly :00, :20, and :40 after the hour.

In https://github.com/pimoroni/enviro/commit/f8616058408be221a04db8b6c138ea2f15d2bae8

closed 2022-09-02 by lowfatcode