pimoroni/enviro #7: Ability to adjust overall and individual wet/dry detection points for Grow sensors

retropangy, 2022-08-10:
With current code sensors have been showing 100 moisture level in almost completely dry pots (to the point of wilting plants) - ability to set wet/dry points and for the scaling to be adjusted appropriately is really needed (similar to Grow Hat functionality).

retropangy, 2022-08-22:
Comment (to myself) - I eventually decided that the easiest thing to do was to split into three different scaled statements and because I can't code I decided to do it in the most hacky way possible whilst not reworking the entire for loop;

`scaled = (min(40, max(0, average - 20)) / 40) * 100`

retropangy, 2022-08-22:
Clicked the wrong button. 🙄

lowfatcode, 2022-08-31:
@retropangy have you ever seen a value other than 100? Have you got three sensors plugged in? Anything else about your setup you can share?

lowfatcode, 2022-09-04:
@retropangy I'm not sure it's a concern that all of the scales map directly on to each other - in fact I don't think it even really makes sense. However the values certainly shouldn't clamp to the extremes as that is useless.

I've extending the min/max range which hopefully means it will now return values for all three of your pots.

There is now a way to setup auto watering thresholds in https://github.com/pimoroni/enviro/commit/5a8646942d78a787be102afca15b144ecff31be8

retropangy, 2022-09-06:
> I've extending the min/max range which hopefully means it will now return values for all three of your pots.

Tried 0.0.8 and it was pegged at 100 on all three. What I've done is make sure the pots are thoroughly "water in the saucer" saturated and changed the min_ms and max_ms values so that the scaled result is just nearly at 100 and I'll monitor the values as they dry out. (min_ms to 50 and max_ms to 105 IIRC). I am having a few issues with the uploads to adafruit_io failing and the board flashing a red warning light - not quite sure what is going on there, don't think it's grow.py at fault though.

ZodiusInfuser, 2022-11-04:
@retropangy Did your changes to the min_ms and max_ms values work? Did the moisture number go down as the plants dried out?