pimoroni/enviro #8: Example code for automatic watering & pump control?

retropangy, 2022-08-10:
Would appreciate example code for handling the pumps in the accessory kit. Ta!

sgtwilko, 2022-08-16:
For anyone wanting to work on this yourself, see this comment about gpio pins to control the pumps:
https://forums.pimoroni.com/t/envirogrow-setup-fails-at-wireless-stage/19975/6

From "hel"
> Schematics for the new things will also be along soon, but if you want to attempt hacking in some pump control in the meantime, I can see from our production schem that they’re connected to GP10, GP11 and GP12. Your code should avoid triggering them at the same time to avoid voltage/current spikes (I seem to remember this was an issue with the OG HAT version of Grow…)

lowfatcode, 2022-09-04:
Ahoy both! Thanks for picking this up. There is auto watering optionally alert tone in https://github.com/pimoroni/enviro/commit/2d722218b8c71d1b94cd9a0ccf19818affec70d4

You can set a target moisture level in `config.py` and either enable or disable auto watering. If auto watering is disabled then Grow will make noise when channels are below the target moisture level. It will beep once, twice, or three times depending on which channel (or if multiple channels need watering each of their beep tones will play one after another with a space between).

This will be rolled into v.0.0.7 later today or tomorrow.

closed 2022-09-04 by lowfatcode