pimoroni/enviro #10: rain sensor

mrgfisher, 2022-08-11:
I've got a working enviro weather for wind speed, direction, temp etc. apart from rain. At present I'm simply logging the payload over http but can see the data just fine:
[image]

Also, when the bucket tips it is waking the device and triggering an upload - I can see the activity LED and a log entry appear a moment later., however (as above) the payload that is sent always says ...  "rain": 0....

I can see in the code where that is set, but what I'm not clear is what to do about it. For example, should/could I add the 'wake reason' to the payload? Then I can use the reason code to increment a rain total concept. I *think* I would then need to add a scenario to the function 'def get_wake_reason():' in board.py.

To my mind, this seems easier than adding to the pico the concept of rain today, and worrying about timezones, the 09:00UTC met definition etc..

If I'm on the right path, happy to write, test and create a PR....

vdomos, 2022-08-15:
Hi,
Same observation for me with this card.
In fact, in the code, the rain data are not currently managed:
in the file **enviro/boards/weather.py**, the function *get_sensor_readings()* always returns the value 0 for the rain sensor:

```python
  return {
    "temperature": round(bme280_data[0], 2),
    "humidity": round(bme280_data[2], 2),
    "pressure": round(bme280_data[1] / 100.0, 2),
    "light": round(ltr_data[BreakoutLTR559.LUX], 2),
    "wind_speed": wind_speed(),
    "rain": 0,                                                              # Always 0 return
    "wind_direction": wind_direction()
  }
```
On the other hand I have not seen how the detection of the rain gauge is managed.
The GPIO pin connected to the rain gauge is declared 

`rain_pin = Pin(10, Pin.IN, Pin.PULL_DOWN)`

 but never used in the code, it seems

peter-mount, 2022-08-16:
I've just discovered the same thing, just had the first bit of rain for weeks and nothing being logged. Looked at the source and noticed it's hard coded to 0.

peter-mount, 2022-08-17:
I've managed to get it recording rain, however to do it I had to effectively rewrite the core of the firmware, sacrificing going into deep sleep as it requires an irq on rain_pin to record a rain drop but the reboot on timeout in the original firmware then wipes out that data, keeping the recorded rain as 0

For me what I've got working is fine as this sensor will  be mains/usb powered however they should get it working in the main firmware.

One I know what I've got is actually working I'll clean it up and put that up on github.

vdomos, 2022-08-17:
I was also thinking about modifying the code to use IRQs, but haven't started yet.
@peter-mount,  I will be very interested in your code.

peter-mount, 2022-08-17:
@vdomos this is working, although it should really have a mutex to prevent a race condition between set_rain() and get_rain()

Also this doesn't work when the pico is rebooted via the timer as that then resets the counter.
It's working for me as I'm not rebooting it between samples.

```
# Rain PIN
rain_pin = Pin(10, Pin.IN, Pin.PULL_DOWN)
rain_count = 0.0
rain_hit = 0

# Maplin bucket sample size in ml https://forums.raspberrypi.com/viewtopic.php?t=263262
# 0.011" according to http://www.philpot.me/weatherinsider.html
rain_sample = 0.2794

# set_rain runs under an IRQ when the rain gauge tips.
# It has a simple anti-bounce routine which will work fine unless it's heavy rain
def set_rain(t):
    global rain_count, rain_hit
    now = time.time()
    if rain_hit == 0 or (now - rain_hit) > 0.3:
        rain_count = rain_count + rain_sample
        rain_hit = now

# IRQ to call set_rain when a drop arrives
rain_pin.irq(lambda pin: set_rain(pin), trigger=Pin.IRQ_FALLING)

# Get rain count and reset
def get_rain():
    global rain_count, rain_hit
    val = rain_count
    rain_count = 0
    return val
```

then replace ``"rain": 0,`` with ``"rain:" get_rain(),`` in get_sensor_readings()

I've had this running all day whilst we had some rare rain falling and it's been working pretty well.

vdomos, 2022-08-18:
thanks @peter-mount , I will test this

RedDogUK, 2022-08-20:
I found some suggested code on the Raspberry Pi project website.  It uses the same rain gauge as the one supplied in the Enviro Weather kit.
[https://projects.raspberrypi.org/en/projects/build-your-own-weather-station/8](url)

mrgfisher, 2022-09-02:
I've done some testing and am reasonably sure that whilst the rain tip will wake the device, there is currently no way to tell upon waking as to why - button, rtc, rain or other. 

@lowfatcode - do you think this bug is something that is fixable in time? Understanding the underlying firmware of the enviro board (i.e. the pimoroni-pico cpp code) is currently beyond me, though once the trigger reason is accessible I'm very happy to contribute to the Python code in this repo.

To be clear, not asking for timescales, just if it is probably fixable. If it isn't I'll pursue use of solar and a battery to keep the board always on.

RedDogUK, 2022-09-03:
> rain tip will wake the device

Hi @mrgfisher, I think the GPIO pin connected to the rain tipper will not wake the device on it's own (using the current script) as the pico is asleep.  The code has to somehow disrupt the RTC alarm, which in turn will wake the pico - but then again I might be wrong as I'm lacking knowledge on this topic.

This is where my coding is stuck.  If I was able to figure out how to disrupt the alarm, I would then be able to get the remaining part of the process to work as intended (incl the wake reason).  Hopefully this will be picked up in an official update in the near future.

lowfatcode, 2022-09-04:
Apologies all, this oversight is fixed in https://github.com/pimoroni/enviro/commit/048a044493842135ec3304111c0f8a1da0ddd5f7

The rainfall sensor has the ability to wake up Enviro Weather so that the time of the event can be logged and the device can go back to sleep. These log entries are stored in `rain.txt` which is then used to calculate the rainfall over the past hour in millimetres when taking a scheduled reading.

The feature will be included in our next release (v0.0.6).

lowfatcode, 2022-09-04:
@mrgfisher thank you for raising an issue for this - please see previous comment, this will be working in v0.0.6 and should be suitable for your solar setup. :-)

closed 2022-09-04 by lowfatcode

mrgfisher, 2022-09-04:
Great news - many thanks @lowfatcode !