pimoroni/enviro #11: Provisioning sometimes crashing after step 4

helgibbons, 2022-08-11:
Provisioning sometimes stalls after entering MQTT/Adafruit IO details, requiring a refresh of the page. Just tried provisioning while connected to Thonny and it looks to be throwing this value error when it happens...

```
2022-08-11 18:07:22 [info     /   90944] > GET /images/logo-weather.svg (200 OK) [243ms]
2022-08-11 18:07:22 [info     /  101248] > GET /images/pirate-logo.svg (200 OK) [302ms]
2022-08-11 18:07:22 [info     /   95632] > GET /images/guide-qr-code.svg (200 OK) [378ms]
2022-08-11 18:07:23 [info     /   88112] > GET /generate_204 (301 Moved Permanently) [195ms]
Task exception wasn't retrieved
future: <Task> coro= <generator object '_handle_request' at 20019fc0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "phew/phew/server.py", line 250, in _handle_request
  File "phew/phew/server.py", line 34, in _parse_query_string
ValueError: need more than 1 values to unpack
2022-08-11 18:08:05 [info     /   99136] > GET /generate_204 (301 Moved Permanently) [275ms]
2022-08-11 18:08:06 [debug    /   94128] > parsed template: enviro/html/header.html (took 9 ms)
2022-08-11 18:08:06 [debug    /   88704] > parsed template: enviro/html/header-mini.html (took 12 ms)
2022-08-11 18:08:06 [debug    /   85872] > parsed template: enviro/html/footer.html (took 80 ms)
2022-08-11 18:08:06 [debug    /   83168] > parsed template: enviro/html/provision-step-4-destination.html (took 467 ms)
```

sgtwilko, 2022-08-16:
I had the same issue.  
If it gets stuck on that page it seems to have lost the previous step values.

When I restarted the process it had the adafruit values but was missing the WiFi details.

helgibbons, 2022-08-16:
If anyone else is running into this issue, you can refresh just the page you're on by clicking on the Enviro banner at the top of the page. If you refresh this way (instead of doing a browser refresh) it seems less likely to lose data entered in previous steps.

robesishere, 2022-08-21:
Same issue here with Adafruit IO

Task exception wasn't retrieved
future: <Task> coro= <generator object '_handle_request' at 20022680>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "phew/phew/server.py", line 250, in _handle_request
  File "phew/phew/server.py", line 34, in _parse_query_string
ValueError: need more than 1 values to unpack

lowfatcode, 2022-08-30:
These issues should be resolved in the latest firmware.

closed 2022-08-30 by lowfatcode

robesishere, 2022-09-06:
Had same issue and output after FW update to 0.0.8 but on second tap of the final configuration button, it completed and began sending data.

Task exception wasn't retrieved
future: <Task> coro= <generator object '_handle_request' at 20027bb0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "phew/phew/server.py", line 248, in _handle_request
  File "phew/phew/server.py", line 34, in _parse_query_string
ValueError: need more than 1 values to unpack