pimoroni/enviro #14: Grow sensors do not report correctly

sgtwilko, 2022-08-12:
Hi,

I've observed two instances of the grow sensors reporting incorrect values:

In this screenshot I had watered only the pot with sensor 1 in it:
[image]
As you can see, sensor 1 stopped reading values, and sensors 2 and 3 showed an increase when there was none.

In this example I watered just pot  with sensor 3 in it:
[image]
As you can see sensor 2 also showed an increase.

I'm wondering if this is to do with the combined loop over all sensors when measuring the times.

ThisIsBenny, 2022-08-16:
I have also sometimes the feeling that wrong datas are reported 😔

Examples where I have some times data reported which total out of the average and not realistic (50 Celsius): 
[image]
[image]

lowfatcode, 2022-08-30:
A new way to measure the moisture sensors is in https://github.com/pimoroni/enviro/commit/9ad7574c197e12e2b35dca2c531eaaec959f7e1e

It now measures each sensor independently rather than in a shared loop that could cause contamination of results.

closed 2022-08-30 by lowfatcode