pimoroni/enviro #17: homekit support

Godley, 2022-08-17:
Hi there,

I've spoken to Paul about implementing HomeKit support - I was just wondering if there's any prior art to this already? The main thing I'm pondering is micro python support and how micro python affects being able to use the "standard" HomeKit python libs. I can't find any micro python libs that just work out of the box, from a brief google.

Gadgetoid, 2022-08-17:
It depends enormously upon the libs but the big three issues with adapting CPython code are:

1. memory constraints
2. memory constraints
3. unimplemented features

Also, is the writing on the wall for Homekit yet? How far long is Matter support... https://csa-iot.org/all-solutions/matter/

Sheesh I just looked into it, and it's very: "First, start by creating a multicast DNS service discovery client." :grimacing:

Godley, 2022-08-17:
I hope the writing's not on the wall..I'm running out of plug sockets for hubs in my utility cupboard.

Think I massively underestimated how chonky the homekit libraries are, e.g [this one](https://github.com/jlusiardi/homekit_python) (to be fair, I did 0 research before committing to particular hardware)...they all seem to have a lot of deps which I doubt there will be a port. I might have to *sigh* set up my own webserver on another device and just send the data to it from enviro. Damn...

Anyway let me know if there's another way 😬

Gadgetoid, 2022-08-17:
It looks like Homebridge and maybe the MQTT plugin could be an answer - https://www.npmjs.com/package/homebridge-mqtt

Definitely not ideal, but would let Enviro continue to be "lightweight" while a Pi (or mini PC) does the heavy lifting.

(I have basically 0 smart home devices :laughing: and don't use any of this stuff, could I possibly be any worse a fit for programming smart home gadgets...)

Godley, 2022-08-17:
my house is basically a phillips hue show home at this point.

Anyway, it now makes sense why every new ecosystem brings a new hub, whether or not it has homekit support. I shall go use homebridge 👍 cheers.

closed 2022-08-17 by Godley