pimoroni/enviro #24: EnviroGrow: solid red LED when using power connector

ExperiMentor, 2022-08-18:
My EnviroGrow works when powering from USB.

But now that I've got a connector for the JST-PH socket, I've connected power that way , it does not seem to boot at all. Rather, if I have my voltage reguklator set anywhere above 1.8v, up to 4.4v, all I get is a solid red on the warning LED. No 'Activity' LED and nothing written to log.txt.

Unlike the other post with red LED who was apparently lacking board firmware, all the programs are present and all works if change back to USB power.

Suggestions?

helgibbons, 2022-08-19:
Do you mean a JST-PH to JST-PH wire? We'd presumed most users would be using the AA/AAA battery pack that's in the kit so wouldn't need additional cables for the battery connector?

Does pressing the reset button after supplying power to the battery connector help? I did notice that I got a red light a few times whilst swapping between different battery situations, but starting the board off with a clean reset seemed to sort it for me.

ExperiMentor, 2022-08-19:
I bought the smaller kit without pumps - so it has no no AA or AAA battery pack.  I'd expected it to work the same off USB power, didn't read that it does not sleep when on USB power, else running off a USB power bank was my plan. Then I was going to connect the power bank to the JST-PH.

No pressing RESET makes no difference. It doesn't even cause a flicker of the red LED

Gadgetoid, 2022-08-19:
It looks like this PR intends to restore "Poke" functionality on USB power - https://github.com/pimoroni/enviro/pull/12- by polling for the onboard button when the board would otherwise enter a blocking wait for the RTC's next event.

In the mean time holding "Poke" and pressing "Reset" should have exactly the same effect as just pressing "Poke".

Indeed the board does not sleep while connected to USB power since this would make it rather hellish to program via Thonny. Connecting your USB battery pack to a vreg and via the JST-PH as you intend should restore sleep functionality.

ExperiMentor, 2022-08-19:
Thanks Gadgetoid.
Your last sentence is what I have been trying. With power supplied via the JST-PH (and nothing connected to the USB), it should operate and sleep as programmed. It is not doing that. Instaed, I just see a solid red LED, and nothing is written to log.txt - showing that the device never woke up.

Gadgetoid, 2022-08-22:
Have you tried with any other JST power supplies? What voltage regulator are you using?

ExperiMentor, 2022-08-22:
I'm using an adjustable voltage regulator with built-in digital voltmeter,
the red LED comes on above about 1.8v, and I can go up to about 4.4v with
that supply.  Or 5.1v with USB battery pack directly. Or 3.4v with an 18600
LiPo cell.
All act the same (solid red LED) and the voltage is arriving on the board
as measured with a multimeter.
Unless there are other ideas to check, I'll need to return the board as
faulty, I think

lowfatcode, 2022-08-30:
The red LED is connected to the onboard RTC chip and the initial state is "on". This suggests that for whatever reason when you're powering from the battery connector that the Pico W isn't powering up at all (since one of the first things it does on power up is tell the RTC to turn the LED off).

Can you confirm that the battery voltage is preset on the VSYS pin of the Pico W?

ExperiMentor, 2022-08-30:
I received a replacement board from your colleagues.
I did check the right voltage was arriving downstream if the JST connector,
but as I've not yet packed it up for return/inspection, I can check the
faulty one if you tell me which pin you mean?

The replacement runs fine on JST power supply, but even though I updated
the firmware, the 2 degree temperature spikes at time of WiFi upload have
returned. See io.adafruit.com/CortmalawHerbs/dashboards/soil-moisture

lowfatcode, 2022-08-31:
Ahh, if the replacement works then that's fine - clearly there is a fault in the original board. Apologies for that, I'm glad our team could sort out a replacement for you. :-)

Those temperature reading spikes are very odd. Are you running the stock firmware?

It looks like you're taking a reading every 15 minutes and uploading every 4 readings?

ExperiMentor, 2022-08-31:
Yes - stock firmware

Yes - reading every 15 min with upload every 4 readings (we worked out that every 4 readings is needed for the io.adafruit free account, because it only allows 30 data points per minute and each upload is 7 data points. I made a change to the timeout programming to delay by 70 seconds instead of 30 if there is overload on this - only happens if uploads have not occurred for some reason)

lowfatcode, 2022-09-02:
Please try the new v0.0.4 firmware that will be released later today - it has a large number of fixes and changes that should help here.

closed 2022-09-02 by lowfatcode

ExperiMentor, 2022-09-02:
Oddly, the temperature spikes have stopped. I didn't change anything, honestly! It's been rebooted a few times but no more than that