pimoroni/enviro #27: Demo code doesn't turn the pumps from the Enviro Grow kit on / off.  Written custom code to resolve issue.

RedDogUK, 2022-08-20:
The published Enviro Grow scripts doesn't have any code to control the three pumps.  I've solved this by writing some custom code.  Details as follows...

Created a new folder on the Pico called "custom".  Created two new files within the custom folder: "pumps.py" and "water_plants.py".

**pumps.py**
`import time
from breakout_bme280 import BreakoutBME280
from breakout_ltr559 import BreakoutLTR559
from machine import Pin, PWM
from enviro.board import i2c

bme280 = BreakoutBME280(i2c, 0x77)
ltr559 = BreakoutLTR559(i2c)

#Water pump GPIO settings
Pump_A = Pin(12, Pin.OUT)
Pump_B = Pin(11, Pin.OUT)
Pump_C = Pin(10, Pin.OUT)

def Pump_1(run):
    Pump_A.value(1) # Turns pump on
    print ('Pump A on')
    time.sleep(run)
    Pump_A.value(0) # Turns pump off
    print ('Pump A off')

def Pump_2(run):
    Pump_B.value(1) # Turns pump on
    print ('Pump B on')
    time.sleep(run)
    Pump_B.value(0) # Turns pump off
    print ('Pump B off')

def Pump_3(run):
    Pump_C.value(1) # Turns pump on
    print ('Pump C on')
    time.sleep(run)
    Pump_C.value(0) # Turns pump off
    print ('Pump C off')

#Testing
#Pump_1(run)
#Pump_2(run)
#Pump_3(run)
`

**water_plants.py**
`import time
import custom
from custom import pumps
import enviro
from enviro.boards import grow

#Pump running time (seconds)
run = 2

#Dry sensor value
dry = 2

#Initiate reading values for testing
reading1 = 0
reading2 = 0
reading3 = 0

#Initial sensor readings
reading1 = round(grow.moisture_readings(2000)[0],2)
reading2 = round(grow.moisture_readings(2000)[1],2)
reading3 = round(grow.moisture_readings(2000)[2],2)

while (reading1 <= dry) or (reading2 <= dry) or (reading3 <= dry):
    
    #reading = enviro.get_sensor_readings()
    reading1 = round(grow.moisture_readings(2000)[0],2)
    reading2 = round(grow.moisture_readings(2000)[1],2)
    reading3 = round(grow.moisture_readings(2000)[2],2)

    print (reading1)
    print (reading2)
    print (reading3)

    if reading1 <= dry:
        pumps.Pump_1(run)
        time.sleep(5)

    if reading2 <= dry:
        pumps.Pump_2(run)
        time.sleep(5)

    if reading3 <= dry:
        pumps.Pump_3(run)
        time.sleep(5)
    
    # For testing
    #reading1 = reading1+1
    #reading2 = reading2+1
    #reading3 = reading3+1

print ('All plants are well watered')
`

I also amended the main.py file as follows:
`import custom
from custom import water_plants`

Inserted just above the enviro.sleep line.
`# water the plants
custom.water_plants`

I hope this helps.

sgtwilko, 2022-08-27:
Nice, can you submit this as a pull request?

RedDogUK, 2022-09-03:
> Nice, can you submit this as a pull request?

@sgtwilko, what do you mean by "pull request"?  My IT jargon dictionary isn't that great.

sgtwilko, 2022-09-03:
Hiya @RedDogUK ,

A pull request is where you start the process to get your code merged into a git repository.

There's many guides online as to how to do this, but I basically boils down to:
You fork ( make a copy of the repo) here on GitHub.

You edit your copy to include the code you want to submit (normally this would be done on a separate branch, but if you're only submitting one thing you don't have to do this step)

You then open a pull request, this is basically you asking to have your code "pulled" and merged into the original repository.

Your pull request would then be reviewed by owner of the repo, and merged if they like it.

I hope that makes sense.

I was tempted to take some of your code and integrate it myself.

lowfatcode, 2022-09-04:
Ahoy both! Thanks for picking this up. There is auto watering optionally alert tone in https://github.com/pimoroni/enviro/commit/2d722218b8c71d1b94cd9a0ccf19818affec70d4

You can set a target moisture level in config.py and either enable or disable auto watering. If auto watering is disabled then Grow will make noise when channels are below the target moisture level. It will beep once, twice, or three times depending on which channel (or if multiple channels need watering each of their beep tones will play one after another with a space between).

This will be rolled into v.0.0.7 later today or tomorrow.

closed 2022-09-04 by lowfatcode

sgtwilko, 2022-09-04:
Hiya @lowfatcode 

Damn, I was just about to submit a PR with auto-watering!
I must have forked just before this was merged.

Looks I've taken a different approach where I check the sensors again after watering.
Want to see my solution?