pimoroni/enviro #38: Add support for more destinations

lowfatcode, 2022-08-30:
A user has requested support for:

WeatherUnderground (https://www.wunderground.com/signup)
Awekas.at (https://www.awekas.at/wp/?lang=en)
Met Office WoW (https://register.metoffice.gov.uk/WaveRegistrationClient/public/register.do?service=weatherobservations)
PWS (https://www.pwsweather.com)
opensensemap (https://opensensemap.org), or
CWOP (http://www.wxqa.com)

lowfatcode, 2022-08-30:
"I would say keep current upload options (but provide more docs) and then add in Wunderground (common option so will be popular, app support, etc), PWS (easy and pretty dashboards) and Met Office or Opensensemap (ties into the maker/community piece)."

Gadgetoid, 2022-09-27:
@MrDrem the relative complexity may be too much for the little RP2040 to handle, not to mention the battery consumption. It might be somewhere between impractical and impossible to pull off.

I wont say never because @lowfatcode is steering this project and has the final word, but I would very much encourage uploading to a local MQTT instance running on a Pi or otherwise, and then running scheduled jobs to copy that data where you want it.

I dare say we could build something to help with this, but one step at a time!