pimoroni/enviro #41: API for extra sensor data/values

lowfatcode, 2022-08-30:
Currently each board type defines a list of sensors it includes but there is no provision to submit extra values (such as sensors connected to the Qw/ST port) alongside these.

I propose adding an optional callback function that can be implemented to add extra values.

Something like this in your `main.py`:

```python
def additional_readings():
  breakout_scd41.start()
    while breakout_scd41.ready() is False:
        time.sleep(5)
  scd41_co2, scd41_temperature, scd41_relative_humidity = breakout_scd41.measure()
  breakout_scd41.stop()

  return {
    "scd41_co2": scd41_co2,
    "scd41_temperature": scd41_temperature,
    "scd41_humidity": scd41_relative_humidity
  }
```

Inspired by https://github.com/pimoroni/enviro/pull/33

fsargent, 2022-08-30:
How can I help fill this out? Would be happy to follow an example of it. 
You'd still need to concat the values into the 'results' payload.

```
reading = enviro.get_sensor_readings()
if additional_readings():
  reading = dict(reading,**additional_readings()) # Extend with additional readings
```

Would you want people to write it into their main.py? Or could use #33 as an example but not accept any more submissions for sensors.

lowfatcode, 2022-08-31:
I must be going mad, I'd already thought of this hence `get_sensor_readings()` followed by `save_reading()`.

The intention was that the user modifies their `main.py` to add any custom values they want to the readings (or indeed to remove values they don't want).

For example to add a custom value into the readings:

```python
breakout_scd41.start()
while breakout_scd41.ready() is False:
  time.sleep(5)
scd41_co2, scd41_temperature, scd41_relative_humidity = breakout_scd41.measure()
breakout_scd41.stop()

reading = enviro.get_sensor_readings()

reading["scd41_co2"] = scd41_co2
reading["scd41_temperature"] = scd41_temperature
reading["scd41_relative_humidity"] = scd41_relative_humidity

enviro.save_reading(reading)
```

or to remove a value that isn't wanted:

```python
reading = enviro.get_sensor_readings()

del reading["humidty"]

enviro.save_reading(reading)
```

This approach seems the most flexible and straightforward - does it fill your usecase @fsargent ?

closed 2022-08-31 by lowfatcode