pimoroni/enviro #62: Wind speed sensor not registering low readings.

MrDrem, 2022-09-04:
I currently go with long periods of the wind speed sensor registering zero for the wind speed (~400 readings with a speed, out of over 5000).

I've currently upped the sensor reading time to 1000ms from 500, and this seems to have improved things considerably, but it may require tweaking up further to record lower wind speeds.

Looking at the equation for working out wind speed, if we only measure for 500ms, and have to get 2 ticks, wind speeds below 0.96 meters per second (2.1 mph) will never be measured.

```
2 ticks in 500ms
average_tick_ms = (ticks[-1] - ticks[0]) / (len(ticks) - 1)
average_tick_ms = 500/1 = 500

rotation_hz = (1000 / average_tick_ms) / 2
rotation_hz = (1000/500)/2 = 2/2 = 1

radius = 7.0
circumference = radius * 2.0 * math.pi
factor = 0.0218  # scaling factor for wind speed in m/s
wind_m_s = rotation_hz * circumference * factor
wind_m_s = 1 * (7*2*pi) * 0.0218 = 0.96 (to 2 decimals)
```

If we redo this for 1 second (1000ms) we get down to a speed of 0.48 m/s (just over 1mph)

If we go to 2 seconds we get down to 0.24 m/s (about half a mile an hour)

With wind speeds these differences are important, we should be able to distinguish between calm and light air (0 m/s and 1 m/s, see https://www.metoffice.gov.uk/weather/guides/coast-and-sea/beaufort-scale ).

This isn't really possible at a reading time of 500ms, but is once you start to go over 1000ms

I've currently set mine to 1500ms, and am getting good low readings now.

lowfatcode, 2022-09-04:
@MrDrem thanks for the feedback.

Perhaps the sample time should be a configuration option. 2mph wind is barely noticable so I think for many users that would be an acceptable floor to their readings.

The compromise is that longer readings mean longer wake times and therefore reduced battery life.

MrDrem, 2022-09-04:
I think that a number of users will be looking to upload to places like the Met Office, and Weather Underground (I've seen requests for this in the upload areas). People who are uploading to such places are generally bothered about low readings, and I think a number of people are likely to use this to replace older Maplin Weather stations that did do readings this low (I'd have to dig though the tweets from https://twitter.com/mitchamweather to find the lowest I recorded with it though)

A config option would be good, maybe set to the mph of the wind that can be detected? I'd suggest that 1000ms as a start would be a good compromise.

lowfatcode, 2022-09-04:
That's compelling enough evidence for me! Updated to 1000ms here: https://github.com/pimoroni/enviro/commit/54cc19876bcc5ff84d4af42d9687501d259a8f02

This change will form part of our v0.0.7 release - thank you! We'll continue to look at ways to better handle this longer term.

closed 2022-09-04 by lowfatcode