pimoroni/enviro #67: Flashing an update .uf2 will erase "config.py" and any logs/pending uploads

Gadgetoid, 2022-09-05:
Since the filesystem .uf2 and full firmware .uf2 both completely erase and replace the filesystem on-device, theyi will also blow away the user config file.

We can fix this - I think - by changing the Pimoroni PicoW Enviro MicroPython build's `_boot.py` (see: https://github.com/micropython/micropython/blob/master/ports/rp2/modules/_boot.py) to set up *two* filesystems.

The "main" filesystem will be first, mounted at "/" and the "config" filesystem would come after, mounted at "/config".

This would let us ship filesystem and firmware .uf2 files that don't rewrite the "config" partition, so user data is preserved.

It would also potentially allow an online or offline tool to create a config .uf2 which could be flashed to a board in lieu of running through the captive portal setup.

In order to do this, however, we need to be able to get discrete "Flash()" instances with specific offsets/sizes. Right now `rp2.Flash()` returns a singleton block device instance that covers the entire filesystem region- https://github.com/micropython/micropython/blob/master/ports/rp2/rp2_flash.c

Gadgetoid, 2022-09-05:
I have opened an Issue to propose changes to `Flash()` (which I will probably raise as a PR, also) so we can make this happen - https://github.com/micropython/micropython/issues/9208

lowfatcode, 2022-09-05:
This sounds like a great idea and I'm all for it. I would suggest that we allow a few more weeks to let the config file format stabilise further since things are changing so fast right now. It would be good to avoid having lots of invalid/mismatched configs out there.

Gadgetoid, 2022-09-05:
Is it worth also adding a `version` to config.py? In future it would let us do a simple check to fail fast on an out of date config.

MrDrem, 2022-09-05:
This would be really helpful, I've found that sometimes pulling and dropping a config file back in doesn't work, as there have been changes to what is being looked for within the config file. Versioning would really help spot this before testing things.

lowfatcode, 2022-09-05:
I guess the `version` would really be a config file version since it may remain stable across many releases. Will have a think about it!

Gadgetoid, 2022-09-12:
Support for flash regions has been merged upstream as of - https://github.com/micropython/micropython/commit/6aa3c946347281875165392c09753547d8c77fc3