pimoroni/enviro #71: Wake reasons not showing in the logs correctly

MrDrem, 2022-09-05:
Looking at lines 211-232 in enviro/__init__.py, I think that the log file should show the reasons why the system has woken from sleep.

Looking at my [log file](https://github.com/pimoroni/enviro/files/9491829/2022-09-05b.log.txt) I can see that most of these are returning 'unknown'.
I know that most of the logs should show 'rtc_alarm' as the cause, but that those at 2022-09-05 18:16:37 and 2022-09-05 19:06:13 should be 'button' as I know those were caused by me hitting the 'poke' button.
I also suspect that at 2022-09-05 18:01:23, 2022-09-05 18:01:26, 2022-09-05 18:01:31, 2022-09-05 18:18:58 & 2022-09-05 18:32:50 should have a wake reason line with 'rain_sensor' against them.

lowfatcode, 2022-09-06:
Is your device powered by USB? If it is then it won't report the wake reason correctly. 

Originally Enviro boards were never intended to be powered via USB when in use - they were meant for battery power where the board fully shuts down between readings.

It seems quite a few people want to run their boards on permanent power via the USB port though so we will add support for that use case in the future.

MrDrem, 2022-09-06:
Nope, most of the time it's running on a [High Capacity Lithium Ion Battery Pack](https://shop.pimoroni.com/products/high-capacity-lithium-ion-battery-pack?variant=32012684591187)

[image]

lowfatcode, 2022-09-06:
Thanks for the feedback, this definitely seems to be a bug - we'll get on it!

ZodiusInfuser, 2022-11-04:
@MrDrem My best guess is that our code reads the button state too late after wake-up so defaults to reporting unknown. There was a similar issue with the Enviro Weather and detecting rain ticks. I will investigate further.

ZodiusInfuser, 2022-11-04:
@MrDrem Just to let you (and others) know. I'm working on fixes for many of the enviro issues over in `https://github.com/pimoroni/enviro/tree/patch`. I believe this is one I have now squashed.

ZodiusInfuser, 2022-11-26:
There is a new release that fixes this issue: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
I'll therefore close this but please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser