pimoroni/enviro #82: How to access local recordings on enviro indoor?

djbloc, 2022-09-10:
When enviro uf2 firmware is loaded, the board is not detected in Thonny 4.0.0 (in Ubuntu) and therefore I can not access the recordings files on the board.

By experimenting, if I nuked the flash and then load the [pimoroni-picow-v1.19.6-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.6/pimoroni-picow-v1.19.6-micropython.uf2) it is detected in Thonny and I can access files. But as soon as the enivro files are loaded and USB is removed and reconnected, the board is not detected and therefore there is no access to the files in Thonny.

Are others able to access local recordings by connecting their board to USB on the computer after going through the environ provision process?

MrDrem, 2022-09-10:
Once you have connected the board, you normally have to hit stop within Thonny (I think because the board autoloads main.py), the board will then show up within the lower right of the screen. I've sometimes had to press reset on the board, and then stop it again to get it to talk correctly.

djbloc, 2022-09-10:
Thank you for the suggestion. Unfortunately hitting stop in Thonny (multiple times) even after pressing the reset button on the board does not seem to work for me on my board. It seems to refuse to detect the board

frederik042853, 2022-09-10:
I have the same issue. Thonny gives the message that the board is busy

diplodocuslongus, 2022-09-30:
And I so do I have the same issue as @djbloc , ubuntu as well. Board doesn't even appear if I do `lsusb`, in Thonny it is not detected (couldn't be if the linux system itself doesn't see the board).

ZodiusInfuser, 2022-11-04:
Thanks for raising this. I am not sure if it the exact cause of the issue described in this thready, but I encountered similar issues with Thonny as you described that seemed to be caused by the addition of battery monitoring code.

Until I push a fix you could try commenting out the below section of code from `enviro/__init__.py`
```python
old_state = Pin(WIFI_CS_PIN).value()
Pin(WIFI_CS_PIN, Pin.OUT, value=True)
sample_count = 10
battery_voltage = 0
for i in range(0, sample_count):
  battery_voltage += (ADC(29).read_u16() * 3.3 / 65535) * 3
battery_voltage /= sample_count
battery_voltage = round(battery_voltage, 3)
Pin(WIFI_CS_PIN).value(old_state)
```
Then add a line like `battery_voltage = 0.0` so that later code still works

ZodiusInfuser, 2022-11-26:
There is a new release that fixes the issue with connecting via Thonny: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
I'll therefore close this but please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser