pimoroni/enviro #83: Adafruit IO Data Uploads - Red LED error light activated (Enviro Weather and Enviro Grow)

RedDogUK, 2022-09-10:
I suspect that the script for uploading data to Adafruit IO causes the red LED on the board to be activated.
[https://github.com/pimoroni/enviro/blob/main/enviro/destinations/adafruit_io.py](url)

I found in my testing of v0.0.8 that the board errored out once the number of upload files reached it's trigger point (as defined in config.py).  I am not certain, but I winder if this is because the current script doesn't have any sleep conditions for when the service suspends the feeds due to the submission limit being reached.  Once this error occurs, I can't access the board via Thonny (Thonny message states "Couldn't find the device automatically..."  The only way I've been able to regain access to the board is to install a fresh version of the firmware - thereby wiping out the stored upload files.

MrDrem, 2022-09-10:
I think that the responses to issue #77 may assist you understanding the red LED. Essentially it's telling you that the upload failed for one reason or another, let it try again and it may go, and that it will be continuing to try whilst it is on.

To get back into the board once it's gone red, connect it, hit the reset button on the back, wait for it to cycle and then press stop within Thonny. That should get you back in.

RedDogUK, 2022-09-11:
@MrDrem, I'm not sure what you are doing differently to me, but I've tried 50+ times (over multiple days) to access the board via Thonny once the red LED has been activated, and I've not been able to connect once.  (connect via USB, use the reset button.)  The only way I've been able to gain access to the board is by putting a fresh version of the firmware onto the board thereby wiping out the upload files.

I'm close to the point of ditching the official firmware and writing my own scripts.