pimoroni/enviro #89: Enviro Urban main.py

tslrkm, 2022-09-12:
Brand new  Enviro Urban and initally updated to:  enviro-v0.0.8.uf2 following the 'easy' [mode method]([url]https://github.com/pimoroni/enviro/blob/main/documentation/upgrading-firmware.md#easy-mode(url))

Rebooted and provisioned went okay.  

Pointing at adafruit.io, where I has an IO+ account.  

Waited for the allotted time period, nothing.  

Pressed POKE - no LED flashing.  

Launched Thonny (version 3.3.13)

Report of log.txt shows only the provisioning setup:

2021-01-01 00:00:02 [error    / 127kB] > missing or corrupt config.py no module named 'config'
2021-01-01 00:00:02 [info     / 125kB] > entering provisioning mode
2021-01-01 00:00:02 [info     / 117kB] > auto detecting board type
2021-01-01 00:00:02 [info     / 116kB]   - urban
2021-01-01 00:00:02 [info     / 114kB] > going into access point mode
2021-01-01 00:00:05 [info     / 112kB]   - 192.168.4.1
2021-01-01 00:00:05 [info     / 110kB] > starting dns server...
2021-01-01 00:00:05 [info     / 108kB] > starting catch all dns server on port 53
2021-01-01 00:00:05 [info     / 106kB] > creating web server...
2021-01-01 00:00:05 [info     / 101kB] > waiting for a client to connect
2021-01-01 00:00:59 [info     /  97kB]   - client connected! (b'f>3\xe9BE',)
2021-01-01 00:00:59 [info     /  95kB] > running provisioning application...
2021-01-01 00:00:59 [info     /  92kB] > starting web server on port 80
2021-01-01 00:01:01 [info     / 118kB] > GET /hotspot-detect.html (301 Moved Permanently) [36ms]
2021-01-01 00:01:02 [info     / 106kB] > GET /wrong-host-redirect (200 OK) [21ms]
2021-01-01 00:01:04 [info     /  92kB] > GET /hotspot-detect.html (301 Moved Permanently) [307ms]
2021-01-01 00:01:04 [info     /  84kB] > GET /wrong-host-redirect (200 OK) [209ms]
2021-01-01 00:01:04 [info     / 114kB] > GET /hotspot-detect.html (301 Moved Permanently) [226ms]
2021-01-01 00:01:05 [debug    /  89kB] > parsed template: enviro/html/header.html (took 18 ms)
2021-01-01 00:01:05 [info     / 104kB] > GET /wrong-host-redirect (200 OK) [193ms]
2021-01-01 00:01:05 [debug    /  92kB] > parsed template: enviro/html/footer.html (took 7 ms)
2021-01-01 00:01:05 [debug    /  90kB] > parsed template: enviro/html/welcome.html (took 337 ms)
2021-01-01 00:01:05 [info     /  87kB] > GET /provision-welcome (200 OK) [701ms]
2021-01-01 00:01:06 [info     /  98kB] > GET /images/pirate-logo.svg (200 OK) [944ms]
2021-01-01 00:01:06 [info     /  91kB] > GET /images/logo-urban.svg (200 OK) [1087ms]
2021-01-01 00:01:06 [info     /  87kB] > GET /images/guide-qr-code.svg (200 OK) [1191ms]
2021-01-01 00:01:06 [info     /  78kB] > GET /hotspot-detect.html (301 Moved Permanently) [344ms]
2021-01-01 00:01:06 [info     /  75kB] > GET /images/photo-urban.jpg (200 OK) [1373ms]
2021-01-01 00:01:07 [debug    /  95kB] > parsed template: enviro/html/header.html (took 25 ms)
2021-01-01 00:01:07 [info     /  88kB] > GET /wrong-host-redirect (200 OK) [194ms]
2021-01-01 00:01:07 [debug    /  85kB] > parsed template: enviro/html/header-mini.html (took 83 ms)
2021-01-01 00:01:07 [debug    / 107kB] > parsed template: enviro/html/footer.html (took 20 ms)
2021-01-01 00:01:07 [debug    / 105kB] > parsed template: enviro/html/provision-step-1-nickname.html (took 356 ms)
2021-01-01 00:01:07 [info     / 102kB] > GET /provision-step-1-nickname (200 OK) [481ms]
2021-01-01 00:01:08 [info     / 106kB] > GET /images/pirate-logo.svg (200 OK) [505ms]
2021-01-01 00:01:08 [info     / 100kB] > GET /images/logo-urban.svg (200 OK) [591ms]
2021-01-01 00:01:08 [info     /  95kB] > GET /images/guide-qr-code.svg (200 OK) [668ms]
2021-01-01 00:01:08 [info     /  91kB] > GET /hotspot-detect.html (301 Moved Permanently) [181ms]
2021-01-01 00:01:08 [info     /  84kB] > GET /wrong-host-redirect (200 OK) [249ms]
2021-01-01 00:01:17 [info     /  79kB] > POST /provision-step-1-nickname (301 Moved Permanently) [790ms]
2021-01-01 00:01:17 [debug    /  93kB] > parsed template: enviro/html/header.html (took 52 ms)
2021-01-01 00:01:17 [debug    /  87kB] > parsed template: enviro/html/header-mini.html (took 11 ms)
2021-01-01 00:01:17 [debug    / 105kB] > parsed template: enviro/html/footer.html (took 14 ms)
2021-01-01 00:01:17 [debug    / 100kB] > parsed template: enviro/html/provision-step-2-wifi.html (took 376 ms)
2021-01-01 00:01:17 [info     /  98kB] > GET /provision-step-2-wifi (200 OK) [492ms]
2021-01-01 00:01:17 [info     /  95kB] > GET /hotspot-detect.html (301 Moved Permanently) [350ms]
2021-01-01 00:01:18 [info     / 112kB] > GET /wrong-host-redirect (200 OK) [252ms]
2021-01-01 00:01:19 [info     /  94kB] > GET /networks.json (200 OK) [1081ms]
2021-01-01 00:01:19 [info     /  75kB] > GET /images/pirate-logo.svg (200 OK) [1505ms]
2021-01-01 00:01:19 [info     / 104kB] > GET /images/logo-urban.svg (200 OK) [1588ms]
2021-01-01 00:01:19 [info     / 100kB] > GET /images/guide-qr-code.svg (200 OK) [1646ms]
2021-01-01 00:01:19 [info     /  95kB] > GET /hotspot-detect.html (301 Moved Permanently) [159ms]
2021-01-01 00:01:20 [info     /  89kB] > GET /wrong-host-redirect (200 OK) [267ms]
2021-01-01 00:02:04 [info     / 101kB] > POST /provision-step-2-wifi (301 Moved Permanently) [197ms]
2021-01-01 00:02:04 [debug    /  70kB] > parsed template: enviro/html/header.html (took 19 ms)
2021-01-01 00:02:04 [debug    /  85kB] > parsed template: enviro/html/header-mini.html (took 11 ms)
[... 26 lines cut ...]
2021-01-01 00:02:42 [info     /  99kB] > GET /provision-step-5-done (200 OK) [605ms]
2021-01-01 00:02:42 [info     /  91kB] > GET /hotspot-detect.html (301 Moved Permanently) [342ms]
2021-01-01 00:02:42 [info     /  77kB] > GET /wrong-host-redirect (200 OK) [135ms]
2021-01-01 00:02:43 [info     /  89kB] > GET /images/pirate-logo.svg (200 OK) [693ms]
2021-01-01 00:02:43 [info     /  83kB] > GET /images/logo-urban.svg (200 OK) [777ms]
2021-01-01 00:02:43 [info     /  79kB] > GET /images/guide-qr-code.svg (200 OK) [843ms]
2021-01-01 00:02:43 [info     /  74kB] > GET /hotspot-detect.html (301 Moved Permanently) [168ms]
2021-01-01 00:02:43 [info     / 109kB] > GET /wrong-host-redirect (200 OK) [67ms]

Pressed RED stop in Thonny and then GREEN start foor main.py.  Following error:

**MicroPython 9dfabcd-dirty on 2022-08-10; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
  File "enviro/__init__.py", line 127, in <module>
OSError: [Errno 22] EINVAL
>>>** 

BTW environment:

WIFI 2.4GHz & 5GHz 
Reporting device - MacBook running Montery (12.5.1.)
Python:  3.7.9 (64 bit)

helgibbons, 2022-09-14:
Not seen that one before! That's an invalid argument error which makes me wonder if the program is trying to use some missing/oddly formatted information - does the contents of your `config.py` look as expected?

tslrkm, 2022-09-15:
Don't think so.  

This is a sanitised version of config.py:

Your booking reference
JFR5WJ5C
You will receive a confirmation email shortl

I’m really loosing patience with my Ender3 V2.  Developed an interesting ‘quirk’ regarding bed adhesion.  If print in the centre  of the bed I get either warping or failed prints.  (With a heated bed or not) but if I print either 30mm forward or back of centre this I get fine prints.  Bed levelled before each print.  

# enviro config file

# you may edit this file by hand but if you enter provisioning mode
# then the file will be automatically overwritten with new details

provisioned = True

# enter a nickname for this board
nickname = 'frome40'

# network access details
wifi_ssid = ‘xxxxxxx'
wifi_password = ‘xxxxxxxxxxxxxx’

# how often to wake up and take a reading (in minutes)
reading_frequency = 5

# where to upload to ("web_hook", "mqtt", "adafruitio")
destination = 'adafruit_io'

# how often to upload data (number of cached readings)
upload_frequency = 1

# web hook settings
custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

# mqtt broker settings
mqtt_broker_address = ''
mqtt_broker_username = ''
mqtt_broker_password = ''

# adafruit ui settings
adafruit_io_username = ‘xxxxx’
adafruit_io_key = ‘xxxxxxxxxxxxxxx’

# influxdb settings
influxdb_org = ''
influxdb_url = ''
influxdb_token = ''
influxdb_bucket = ''

# grow specific settings
auto_water = False
moisture_target_1 = 50
moisture_target_2 = 50
moisture_target_3 = 50

My first thought was the config.py was screwed up so I re-provisioned it.  

I also altered the values for the reading and upload frequencies for debug purposes.  If the thing had actually worked I'd have re-edited these to something more sane.

helgibbons, 2022-09-15:
Hello! Yeah, your config file looks OK to me. Sorry to hear about your bed adhesion issues - I've found Prusa printers to be very drama free!

Your error seems to be pointing at the bit of code that tries to identify which board you have. It looks like it manages to identify your Urban correctly from the logs though - do you get that same error every time you try to run `main.py` through Thonny? Does it try to go back into provisioning mode (with the rapidly flashing light) every time?

tslrkm, 2022-09-15:
> Hello! Yeah, your config file looks OK to me. Sorry to hear about your bed adhesion issues - I've found Prusa printers to be very drama free!
>
> Your error seems to be pointing at the bit of code that [...]

Yeah it will reset into provisioning mode when holding down POKE & RESET as in the troubleshooting guide.  I am starting to think I may have a hardware issue here.

helgibbons, 2022-09-15:
I'd maybe be tempted to try [clearing the flash](https://github.com/pimoroni/enviro/blob/main/documentation/upgrading-firmware.md#starting-from-scratch) and then copying the enviro image over again, just in case there's some old files lurking around messing things up somehow?

But, if that doesn't help, yeah - the weird error message and the fact that it relates to the figuring out what kind of board you have makes me think that there might be something awry with the hardware - if you drop an email to support@pimoroni.com with a link to this issue they should be able to sort you out with a swap.

tslrkm, 2022-09-15:
Well I've give Flash_nuke a go and see what happens...  

https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory

then I'll layer down the enviro-0.0.8.uf2 once more.

tslrkm, 2022-09-16:
Right well flash nuke used then re applied v0.0.8.  Tested overnight and seems now to be working, so that must be it.  Thanks for the help.

closed 2022-09-16 by tslrkm

helgibbons, 2022-09-16:
Oh good stuff - let us know if you have further issues!