pimoroni/enviro #91: Warning Light Flashing After Upgrade to 0.0.8

TooManyEggrolls, 2022-09-13:
After upgrading enviro indoor to 0.0.8, the warning light consistently flashes.  This appears to happen at the time of upload to Adafruit IO.  Collection rate every 15min, upload after 5 readings, running on battery power (3 AAs).  enviro-v0.0.8.uf2 was used to upgrade firmware from stock version, presumably 0.0.2.  I can see that my data is uploaded to Adafruit IO.  The warning light never stops flashing once its starts flashing.

helgibbons, 2022-09-14:
If you're on a free adafruit.io plan you might be exceeding the 30 data points a minute upload limit - try uploading every set of 3 readings rather than every 5?

TooManyEggrolls, 2022-09-15:
I re-provisioned it (uploaded same firmware again) to start "fresh".  I used every 15min and upload after 1 collection, should be better than 3, right? I WFH so it sat next to me all day. I did this b/w 7-8am CT today 9/14. All is fine all day. Left my desk around 4:15pm ish. Came back down around 5:45pm and it's flashing red.  Hit the poke button and it stops.  Came back down to desk again 2hrs later , flashing red again.  The upgrade for sure broke things. And it's right next to router, so signal not an issue at all. 

 Wondering if I should try Influx Cloud since I have a lot of experience with Influx OSS.

helgibbons, 2022-09-15:
Ah yeah, sounds like you might be running into a bug with v0.0.8 then. Anything in the logs to show what might have happened?

Anecdotally, my devices posting into Influx (either cloud or OSS) have been well behaved so far (though I've seen random crashes / lapses in data with devices using v0.0.8 and adafruit.io and MQTT endpoints).

jamesjarvis, 2022-09-20:
I've seen a strange bug where the RTC synced with a date that is 2 months in the future, so all uploads to adafruit.io failed with `MQTT ERROR: error saving data to X/groups/enviro, data created_at may not be in the future`. This then blocked the upload of any future records as well.

Not sure if this RTC error is related to #79 at all, but it fixed itself once turned off and on again

logs:

```
2022-11-20 18:00:00 [debug    / 106kB]   - reset
2022-11-20 18:00:02 [debug    / 115kB] > performing startup
2022-11-20 18:00:02 [info     / 127kB]   - wake reason: external_trigger
2022-11-20 18:00:02 [debug    / 125kB]   - turn on activity led
2022-11-20 18:00:02 [debug    / 122kB]   - starting sensor
2022-11-20 18:00:02 [debug    / 120kB]   - wait 5 seconds for airflow
2022-11-20 18:00:07 [debug    / 101kB]   - taking pms5003i reading
2022-11-20 18:00:08 [info     /  96kB] > 53 cache files need uploading
2022-11-20 18:00:08 [info     /  93kB] > connecting to wifi network 'AstraZeneca_Tracking_Chip_93fd83'
2022-11-20 18:00:10 [debug    /  87kB]   - connecting
2022-11-20 18:00:13 [debug    / 120kB]   - connecting
2022-11-20 18:00:15 [info     / 111kB]   - ip address:  192.168.1.106
2022-11-20 18:00:16 [error    /  94kB] ! failed to upload '2022-11-19T08:50:43Z.json' to adafruit_io
2022-11-20 18:00:16 [error    /  92kB] ! reading upload failed
2022-11-20 18:00:16 [info     /  90kB] > going to sleep
2022-11-20 18:00:16 [debug    /  88kB]   - clearing and disabling timer and alarm
2022-11-20 18:00:16 [info     / 124kB]   - setting alarm to wake at 18:15pm
2022-11-20 18:00:16 [info     / 108kB]   - shutting down
2022-11-20 18:00:16 [debug    / 106kB]   - on usb power (so can't shutdown) halt and reset instead
2000-01-01 00:00:02 [debug    / 115kB] > performing startup
2000-01-01 00:00:02 [info     / 126kB]   - wake reason: external_trigger
2000-01-01 00:00:02 [debug    / 124kB]   - turn on activity led
2000-01-01 00:00:02 [info     / 122kB] > clock not set, synchronise from ntp server
2000-01-01 00:00:02 [info     / 120kB] > connecting to wifi network 'AstraZeneca_Tracking_Chip_93fd83'
2000-01-01 00:00:04 [debug    / 114kB]   - connecting
2000-01-01 00:00:07 [debug    / 100kB]   - connecting
2000-01-01 00:00:09 [info     /  92kB]   - ip address:  192.168.1.106
2022-09-20 18:15:28 [info     /  89kB]   - rtc synched
2022-09-20 18:15:28 [debug    / 128kB]   - starting sensor
2022-09-20 18:15:28 [debug    / 126kB]   - wait 5 seconds for airflow
2022-09-20 18:15:33 [debug    / 106kB]   - taking pms5003i reading
2022-09-20 18:15:34 [info     / 106kB] > 54 cache files need uploading
2022-09-20 18:15:34 [info     / 103kB] > already connected to wifi
2022-09-20 18:15:35 [info     / 120kB]   - uploaded 2022-09-20T18:15:34Z.json to adafruit_io
2022-09-20 18:15:35 [error    / 106kB] ! failed to upload '2022-11-19T08:50:43Z.json' to adafruit_io
2022-09-20 18:15:35 [error    / 104kB] ! reading upload failed
2022-09-20 18:15:36 [info     / 102kB] > going to sleep
2022-09-20 18:15:36 [debug    / 100kB]   - clearing and disabling timer and alarm
2022-09-20 18:15:36 [info     /  97kB]   - setting alarm to wake at 18:30pm
2022-09-20 18:15:36 [info     /  95kB]   - shutting down
2022-09-20 18:15:36 [debug    /  93kB]   - on usb power (so can't shutdown) halt and reset instead
```

MrDrem, 2022-09-21:
So looking at your log file, you've a whole load of old log files (~53 of them I think from reading this line: `2022-11-20 18:00:08 [info     /  96kB] > 53 cache files need uploading`) that haven't uploaded previously, most likely because they have the wrong dates in them (from this line: `2022-11-20 18:00:16 [error    /  94kB] ! failed to upload '2022-11-19T08:50:43Z.json' to adafruit_io`).

These will be in on of the directories on the device that you can see when you connect it via USB.

I'm not sure what the cause of the wrong time is likely to be, it's certainly correct on this run (from this line: `2022-09-20 18:15:35 [info     / 120kB]   - uploaded 2022-09-20T18:15:34Z.json to adafruit_io`), and was corrected here after you reset the device `2022-09-20 18:15:28 [info     /  89kB]   - rtc synched`. I would guess that it was wrong from the start of that run, as the RTC is only set when the device starts, so ~13 hours before you restarted it this time (If I'm right about the files).

I don't think that this is related to #79 as in that the RTC stops and the device stops waking, I suspect in this case the time was wrong at the start (or something caused the RTC to speed, but that seems unlikely). You could check by looking at the 53 file names I think.

nopuk, 2022-09-21:
I have a similar issue after 'upgrading' to 0.0.8 (using easy method). Red light is flashing, log file says failed to upload to local server - so its not just an Adafruit issue?