pimoroni/enviro #92: Same timestamp used for all intermediate records from influxdb.py

adam-macgill, 2022-09-13:
All the records over period get the same submission time as '_time' in influxdb rather than the time from the 'readings' log. (well one second apart)

In influxdb.py the timestamp looks to be extracted from 'readings' and formatted correctly but I can't see where it is put back into the 'data' to be uploaded.

It may be an issue with influx 2.4 (which I am using) ignoring the field. I will runs as one recording -> one post for now.

MrDrem, 2022-09-13:
Is that not because the submissions do all happen at virtually the same time if you are uploading in a batch? The reading time should be stored as a separate timestamp that should also be available to you.

I've not played with influxdb yet, I'm trying to get mine stable with io.adafruit.com first.

helgibbons, 2022-09-13:
@adam-macgill This is fixed in https://github.com/pimoroni/enviro/commit/d8da4d3606c492769443e274d0a99964df8c1ea5 - if you edit your `influxdb.py` to include that change the timestamps should behave themselves.

MrDrem, 2022-09-13:
@helgibbons can I send data to two locations? 

I'd like to send it to a test influxdb to play with, whist continuing to send it to adafruit so I can keep on seeing what's going on as I do at the moment easily.

adam-macgill, 2022-09-13:
> @adam-macgill This is fixed in [d8da4d3](https://github.com/pimoroni/enviro/commit/d8da4d3606c492769443e274d0a99964df8c1ea5) - if you edit your `influxdb.py` to include that change the timestamps should behave themselves.

Thanks - that did the job. :-)

closed 2022-09-13 by adam-macgill

helgibbons, 2022-09-14:
Awesome!

@MrDrem - to do that you'd need to splice the contents of `influxdb.py` into your `adafruit_io.py` (and probably specify the url in there as well, as `config.py` only allows for one endpoint URL). Might be fiddly as it looks like the two endpoints require slightly differently formatted data?

I'm enjoying it so far - tons of customisation options if you host your own setup (or the upload limits for the free Cloud version seem quite generous).