pimoroni/enviro #95: Warning light is always on.

keith2146, 2022-09-16:
Hi, 
This morning I found that the warning light (Enviro - weather) was on all the time. I pressed the Reset and Poke buttons to restart it, but it didn't work. I tried to connect the pico to USB and use Thonny to check. However, I can't connect to the pico either. I have tried using battery and USB power supply, both have the same problem. 
Is there any other solution? Thanks.

ZodiusInfuser, 2022-11-11:
Hi, were you able to resolve this yourself, or are you still encountering the issue? There's a new build I am working on in the `patch` branch that may have resolved this

keith2146, 2022-11-17:
Hi, I haven't solved this problem yet, do you have any suggestions? 
Thanks.

ZodiusInfuser, 2022-11-17:
Do you recall which version of the Enviro software you have on your board? The old v0.0.2 used to keep readings files around, so would need them to be regularly deleted in order to prevent the memory filling up. That doesn't explain the Thonny issues though, which I know were introduced by v0.0.8.

One thing to try is whilst connected to USB, try using the reset on the board then try connecting via Thonny. If that doesn't work then all I can really suggest to you is to reflash the board, and try out the new code that is over in this branch: https://github.com/pimoroni/enviro/tree/patch. This will mean you'll loose all your non-uploaded reading though, but it sounds like you cannot access them anyway.

ZodiusInfuser, 2022-11-26:
There is a new release that may fix this issue: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
I'll therefore close this but please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser