pimoroni/enviro #97: Stopped uploading and pulsing working light

nopuk, 2022-09-21:
Hello,
My enviro weather was working fine, then after about 8 hrs the white pulsing light just pulses until I go and hit reset or do something - it stopped transmitting data in this time too. This has happened two nights in a row and below is my log file . At 19:40 is when the pulsing started and continued till I took the unit down (20 hours later).
`2022-09-18 19:25:02 [info     /  116016] hey enviro, let's go!
2022-09-18 19:25:02 [info     /  114320]  - --=-=-===-=-=-- -
2022-09-18 19:25:02 [info     /  112720]
2022-09-18 19:25:03 [debug    /  111104] > performing startup
2022-09-18 19:25:03 [debug    /  109408]   - hold vsys_en high
2022-09-18 19:25:03 [info     /  107632]   - wake reason: unknown
2022-09-18 19:25:03 [debug    /  105904]   - turn on activity led
2022-09-18 19:25:03 [debug    /  101440] > 89 blocks free out of 212
2022-09-18 19:25:04 [info     /   92320] > 1 cache files need uploading
2022-09-18 19:25:04 [info     /  115792] > connecting to wifi network 'asus_monster_mesh'
2022-09-18 19:25:09 [error    /   95664] ! failed to connect to wireless network asus_monster_mesh
2022-09-18 19:25:09 [error    /   93520]   - cannot upload readings, wifi connection failed
2022-09-18 19:25:09 [info     /   91712] > going to sleep
2022-09-18 19:25:10 [debug    /   89664]   - clearing rtc alarm flags
2022-09-18 19:25:10 [info     /   87632]   - setting timer to wake in 5 minutes
2022-09-18 19:25:10 [info     /   85696]   - shutting down
2022-09-18 19:30:02 [info     /  119424]
2022-09-18 19:30:02 [info     /  117808] hey enviro, let's go!
2022-09-18 19:30:02 [info     /  116112]  - --=-=-===-=-=-- -
2022-09-18 19:30:02 [info     /  114512]
2022-09-18 19:30:03 [debug    /  112896] > performing startup
2022-09-18 19:30:03 [debug    /  111200]   - hold vsys_en high
2022-09-18 19:30:03 [info     /  109424]   - wake reason: unknown
2022-09-18 19:30:03 [debug    /  107696]   - turn on activity led
2022-09-18 19:30:03 [debug    /  103232] > 89 blocks free out of 212
2022-09-18 19:30:04 [info     /   94032] > 2 cache files need uploading
2022-09-18 19:30:04 [info     /  115840] > connecting to wifi network 'asus_monster_mesh'
2022-09-18 19:30:09 [info     /   96048]   - ip address:  192.168.2.219
2022-09-18 19:30:10 [info     /  110976] > uploading cached readings to 192.168.2.39
2022-09-18 19:30:10 [info     /  104720]   - uploaded 2022-09-18 19:25:04.json
2022-09-18 19:30:10 [info     /  100144]   - uploaded 2022-09-18 19:30:04.json
2022-09-18 19:30:10 [info     /   98208] > going to sleep
2022-09-18 19:30:10 [debug    /   96336]   - clearing rtc alarm flags
2022-09-18 19:30:10 [info     /   94304]   - setting timer to wake in 5 minutes
2022-09-18 19:30:10 [info     /   92368]   - shutting down
2022-09-18 19:35:02 [info     /  118528]
2022-09-18 19:35:02 [info     /  116912] hey enviro, let's go!
2022-09-18 19:35:02 [info     /  115216]  - --=-=-===-=-=-- -
2022-09-18 19:35:02 [info     /  113616]
2022-09-18 19:35:02 [debug    /  112000] > performing startup
2022-09-18 19:35:03 [debug    /  110304]   - hold vsys_en high
2022-09-18 19:35:03 [info     /  108528]   - wake reason: unknown
2022-09-18 19:35:03 [debug    /  106800]   - turn on activity led
2022-09-18 19:35:03 [debug    /  102336] > 88 blocks free out of 212
2022-09-18 19:35:04 [info     /   93232] > 1 cache files need uploading
2022-09-18 19:35:04 [info     /  115840] > connecting to wifi network 'asus_monster_mesh'
2022-09-18 19:35:09 [info     /   97280]   - ip address:  192.168.2.219
2022-09-18 19:35:09 [info     /  110976] > uploading cached readings to 192.168.2.39
2022-09-18 19:35:09 [info     /  104896]   - uploaded 2022-09-18 19:35:04.json
2022-09-18 19:35:10 [info     /  102784] > going to sleep
2022-09-18 19:35:10 [debug    /  100912]   - clearing rtc alarm flags
2022-09-18 19:35:10 [info     /   98880]   - setting timer to wake in 5 minutes
2022-09-18 19:35:10 [info     /   96944]   - shutting down
2022-09-18 19:40:02 [info     /  117792]
2022-09-18 19:40:02 [info     /  116176] hey enviro, let's go!
2022-09-18 19:40:02 [info     /  114480]  - --=-=-===-=-=-- -
2022-09-18 19:40:03 [info     /  112880]
2022-09-18 19:40:03 [debug    /  111264] > performing startup
2022-09-18 19:40:03 [debug    /  109568]   - hold vsys_en high
2022-09-18 19:40:03 [info     /  107792]   - wake reason: unknown
2022-09-18 19:40:03 [debug    /  106064]   - turn on activity led
2022-09-18 19:40:03 [debug    /  101600] > 88 blocks free out of 212
`

ZodiusInfuser, 2022-11-11:
Hi. Do you know which version of the firmware you were running? The white LED pulsing is usually an indication that the board is attempting to upload a file. The fact you are not seeing that at the end of the log is strange.

There have been a bunch of issues with Enviro Weather that I've been attempting to solve over in the `patch` branch. Perhaps its worth you trying out that code and reporting back?

nopuk, 2022-11-11:
Hello,
This was experienced with v008, I tried 002 and while it worked better it
still never managed more than a few days at best. In both cases the device
is just not reliable. Sadly I think I will attempt to build (using a guide)
a raspberry pi weather station in the hope it might be a bit more reliable.
Honeslty kinda annoyed that pimoroni sell the board as a product without
mentioning anything on the webpage.

Patrick