pimoroni/enviro #99: Wind and Rain sensor 'freeze' the board

L0cutus, 2022-09-28:
Hello, 
Weather Station kit  with Pico W Aboard and latest 0.08 firmware (but the same thing happens with the original one), if i run the board WITHOUT the RJ11 connector inserted, the board work as expected with mosquitto, if i only insert one or all the RJ11 connector, it works for some time (mins or hours) and then it stops working (batteries are ok).
Any idea or suggestions?

Thanks!

MrDrem, 2022-09-28:
The rainfall sensor interrupting the wake alarms is a known bug - see #70 

I had thought that there were issues with the wind sensors, but I now suspect that it's an endpoint bug, that may also cause issues with the RTC, see issue #79

L0cutus, 2022-09-28:
i'm not a programmer (i'm only a beginner with python and C/C++) so i think i cannot help very much :-)
Do you know if does exists the same software but written in Arduino IDE C/C++?
Do you think the problem is in the program logic or in the micropython os?
Thanks!

MrDrem, 2022-09-28:
For the first, I'm fairly sure that it's a coding bug, but my amateur attempts at fixing it didn't work. I believe from comments elsewhere that the lead from Pimoroni is back next week, and I hope that this is a fairly easy fix.

The issues around the RTC and sending things to endpoints that time out and other issues I again suspect to be software, as it was working better with io.adafruit.com with the initial 0.0.2 version. Also given that it works fine with a local influxDB instance that obviously has a faster connection and no rate limiting.

These are however just observations, and should as such be taken with the appropriate amount of salt.

L0cutus, 2022-09-28:
yes i've just read the issue #86 and just updated my installation, only connected the wind RJ11 and waiting for tonight test...
i'm using  infludb + mosquitto + telegraph on my raspberry pi, an update every 5min and upload every 2 readings.
For now thanks for help!

L0cutus, 2022-09-29:
Unfortunately the keepalive fix doesn't solve the problem for me :-/
i have attached the log
[log.txt](https://github.com/pimoroni/enviro/files/9671384/log.txt)

daved38, 2022-10-18:
I've not seen any bug fixes forthcoming for the rain senor issues. Rain measurement in my view is fundermental for any outdoor weather station.

I have suffered the same issue and attach the log file
[log.txt](https://github.com/pimoroni/enviro/files/9810258/log.txt)

and rain.txt for reference
[rain.txt](https://github.com/pimoroni/enviro/files/9810267/rain.txt)

One observation I do have is when I bring the the Enviro Weather (after disconnecting sensors!) and connect via USB the following happens:

[image]

I have written to  support@pimoroni.com - "hopefully there'll be an update to the bug forthcoming"

ZodiusInfuser, 2022-11-11:
The issues with the rain sensor should now be fixed in the `patch` branch.

ZodiusInfuser, 2022-11-26:
There is a new release that should fix this issue, so I'm closing this: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
Please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser