pimoroni/enviro #100: Grow making unexpected tweeting noises

ExperiMentor, 2022-09-29:
I have EnviroGrow with no water pumps and the new AutoWater feature set to "off" in config.py. 
Since upgrading to v0.0.8, my wife is complaining of occasional brief 'tweeting' noises "like a metallic bird". I cannot hear anything myself. Trying to track it down:
1. It occurred at a time with sensor measurement at which there was no data upload to adafruit_io
2. It occurred immediately on connection to Thonny by USB
3. It MAY have stopped when I changed config.py to have moisture_target values set to zero instead of the default 50, ie 

# grow specific settings
auto_water = False
moisture_target_1 = 0
moisture_target_2 = 0
moisture_target_3 = 0

helgibbons, 2022-09-29:
That's the 'needs watering' buzzer alerts - they'll trigger when a moisture sensor reading is below the 'moisture_target` number, and they make a different metallic bird noise for each channel - one bleep for channel 1, two bleeps for channel 2 etc

Setting the moisture targets to 0 should stop them from happening - maybe a more obvious 'buzzer alerts on' setting in `config.py` might be useful @lowfatcode ?

ExperiMentor, 2022-09-29:
Thanks, that makes sense!  Mine must be complaining about Channel 3, which has no sensor attached at present.

However, @lowfatcode rather than a 'more obvious 'buzzer alerts on' setting in config.py ' it may make more sense to simply document that the target should be set to zero for channels that are not in use or buzzer not required (or default them all to zero anyway?)

helgibbons, 2022-09-29:
I'm working on a learn article for Grow at the moment, so that should hopefully help to make this behaviour a bit less mysterious :)

ZodiusInfuser, 2022-11-26:
There is a new release that adds in an extra provisioning webpage that lets the moisture targets be set, and includes a note about the chirping noise: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
I'm therefore closing this. Thanks

closed 2022-11-26 by ZodiusInfuser