pimoroni/enviro #101: Calibrating Temperature, Pressure, Humidity scales (Grow, and others?)

ExperiMentor, 2022-09-29:
There was previous discussion that ability to calibrate readings would be useful, as the raw readings can be some way off. 

Presently, I';ve tweaked enviro/boards/grow.py as below, but it would surely be better done in config.py (and similar likely applies to ther boards?)

def get_sensor_readings():
  # bme280 returns the register contents immediately and then starts a new reading
  # we want the current reading so do a dummy read to discard register contents first
  bme280.read()
  time.sleep(0.1)
  bme280_data = bme280.read() 
  **adj_temp = bme280_data[0] - 2.0
  adj_press = (bme280_data[1] / 100.0) + 5.4
  adj_humidity = bme280_data[2] + 11.0**

  ltr_data = ltr559.get_reading()

  moisture_levels = moisture_readings()
  
  water(moisture_levels) # run pumps if needed

  from ucollections import OrderedDict

  return OrderedDict({
    **"temperature": round(adj_temp, 2),
    "humidity": round(adj_humidity, 2),
    "pressure": round(adj_press, 2),**
    "luminance": round(ltr_data[BreakoutLTR559.LUX], 2),

ZodiusInfuser, 2022-11-04:
Thanks for suggesting this! I assume the `2.0`, `5.4` etc are your offset values in the above code?

ExperiMentor, 2022-12-09:
Yes, -2.0, +5.4 and +11.0 [note: signs are needed] are the values I found helpful, but would be better placed in config.py