pimoroni/enviro #102: No wifi connection v0.0.7+

flippest, 2022-09-29:
Upgraded from v0.0.2 to v0.0.7 and wifi no longer connects.  Tested several configurations of 2.4Ghz vs 5Ghz to no avail. Downgraded to v0.0.2 to confirm wifi hardware functionality and was able to connect.  Flashing back to v0.0.7 as well as upgrading to v0.0.8 wifi connectivity issue persists

MPY: soft reboot
       ___            ___            ___          ___          ___            ___       
      /  /\          /__/\          /__/\        /  /\        /  /\          /  /\      
     /  /:/_         \  \:\         \  \:\      /  /:/       /  /::\        /  /::\     
    /  /:/ /\         \  \:\         \  \:\    /  /:/       /  /:/\:\      /  /:/\:\    
   /  /:/ /:/_    _____\__\:\    ___  \  \:\  /__/::\      /  /:/~/:/     /  /:/  \:\   
  /__/:/ /:/ /\  /__/::::::::\  /___\  \__\:\ \__\/\:\__  /__/:/ /:/___  /__/:/ \__\:\  
  \  \:\/:/ /:/  \  \:\~~~__\/  \  \:\ |  |:|    \  \:\/\ \  \:\/:::::/  \  \:\ /  /:/  
   \  \::/ /:/    \  \:\         \  \:\|  |:|     \__\::/  \  \::/~~~`    \  \:\  /:/   
    \  \:\/:/      \  \:\         \  \:\__|:|     /  /:/    \  \:\         \  \:\/:/    
     \  \::/        \  \:\         \  \::::/     /__/:/      \  \:\         \  \::/     
      \__\/          \__\/          `~~~~~`      \__\/        \__\/          \__\/      

    -  --  ---- -----=--==--===  hey enviro, let's go!  ===--==--=----- ----  --  -     

2022-09-29 18:07:52 [debug    / 124kB] > performing startup
2022-09-29 18:07:52 [info     / 122kB]   - wake reason: unknown
2022-09-29 18:07:52 [debug    / 120kB]   - turn on activity led
2022-09-29 18:07:52 [info     / 113kB] > 4 cache files need uploading
2022-09-29 18:07:52 [info     / 111kB] > connecting to wifi network 'Redacted'
2022-09-29 18:08:13 [debug    / 108kB]   - connecting
2022-09-29 18:08:43 [error    /  94kB] ! failed to connect to wireless network Redacted
2022-09-29 18:08:43 [error    /  91kB] ! reading upload failed
2022-09-29 18:08:43 [info     /  89kB] > going to sleep
2022-09-29 18:08:43 [debug    /  87kB]   - clearing and disabling timer and alarm
2022-09-29 18:08:43 [info     /  85kB]   - setting alarm to wake at 18:10pm
2022-09-29 18:08:43 [info     /  83kB]   - shutting down
2022-09-29 18:08:43 [debug    / 126kB]   - on usb power (so can't shutdown) halt and reset instead

ZodiusInfuser, 2022-11-04:
Thanks both for raising this. I am not sure of the exact issue that would cause this yet, but I have been experiencing WiFi issues that were caused by the reading of the battery voltage, particularly when on USB power.

If you are still interested in diagnosing this, could you try commenting out these lines from `enviro/__init__.py`
```python
old_state = Pin(WIFI_CS_PIN).value()
Pin(WIFI_CS_PIN, Pin.OUT, value=True)
sample_count = 10
battery_voltage = 0
for i in range(0, sample_count):
  battery_voltage += (ADC(29).read_u16() * 3.3 / 65535) * 3
battery_voltage /= sample_count
battery_voltage = round(battery_voltage, 3)
Pin(WIFI_CS_PIN).value(old_state)
```
Then add a line like `battery_voltage = 0.0` so that later code still works

ZodiusInfuser, 2022-11-26:
There is a new release that temporarily fixes this by disabling battery monitoring: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
Please could you update and retest?