pimoroni/enviro #103: json files can't be copied off of the boards to Windows machines

MrDrem, 2022-09-30:
My Wifi failed yesterday, which meant that I had a whole load of json files that failed to upload.

I tried to grab them over to my windows laptop, but because the file names had ':' in them, I couldn't, which was a shame as I ended up dropping them to get things working again. Could the ':' be changed to another character?

See https://learn.microsoft.com/en-us/windows/win32/fileio/naming-a-file for 'reserved' characters.

ZodiusInfuser, 2022-11-04:
Thanks for raising this. That sounds easy enough to fix. Just need to decide on substitute characters, as my understanding is the filename is currently following a particular standard for logging times.

ZodiusInfuser, 2022-11-10:
I have added a fix for this to the `patch` branch. Changed the `:` to `_`.

ZodiusInfuser, 2022-11-26:
There is a new release that should fix this issue, so I'm closing this: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
Please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser