pimoroni/enviro #104: Weather v0.0.8 MQTT unable to work a full day

nopuk, 2022-09-30:
Hello,
Parts of this issue seem to have been raised a few times, such as #84 but it does not include everything. So I thought I would summarize my issues:
Enviro weather using v0.0.8 - unable to last more than 7 working hours. It reports fine, then starts missing connections and eventually just tries to upload and hangs (as in the log file says attempting to connect and nothing else) and the white LED puless for hours until stopped.  A poke gets it to upload again, but the cycle repeats. Setting the Keepalive=10 does help but only in getting it to work for slightly longer times.
As a weather station unit its basically unable to preform its job without constant supervision.

jctooley, 2022-10-04:
Two questions on your setup. 

1. Do you run the board from a battery or usb?

2. Does your MQTT broker require a username and password for a client to connect?

With regards the issue with the keep alive I found it wouldn't connect at all without it set. 

Mine current is running from a battery with no username/password set. It runs happily until the battery is flat. This is in my office so no rain is being triggered which is another issue.

nopuk, 2022-10-04:
Hello,
1) Runs from battery, 2x AAA Ni 1.2v, 750mAH (recharge)
2) No password or username needed

Mind would run for a bit without keep alive, but even with that it just
seems to give up after a few hours (the longest streak I have recorded is 7
). The symptom is the board just pulses white and the log files its started
to connect and thats it, just seems to hang. The board speaking to my
router had this issue, I then installed a wifi range extender right next to
it, hoping it was a signal strength issue, but the problem persists. Very
odd that you do not seem to experience this issue.

ZodiusInfuser, 2022-11-26:
There is a new release that adds the keepalive parameter back. @nopuk could you update and retest please?