pimoroni/enviro #109: Enviro Indoor stops posting via MQTT when storage is full.

maddogeco, 2022-10-26:
I have a pair of Enviro Indoor’s (Pico W boards). I have set them up to post every 5 minutes to Home Assistant via MQTT. They were posting every 5 minutes like little champs for 3 weeks. After 3 weeks or so they stop posting. When I plug them into Thonny I can see there are lots of reading log files and no free disk space. 
Once I delete the log files, they will continue posting readings until all the storage space is full again. 
Can the log files be automatically removed after a set time or remove the oldest as the disk fills up.

ZodiusInfuser, 2022-11-04:
Thanks for raising this. I will look into the best way to do this.

ZodiusInfuser, 2022-11-15:
@maddogeco Do you recall which log files specifically? Checking the most recent code, the main `log.txt` auto truncates, and files in `uploads` are deleted after they are uploaded. The only files that are not deleted are the ones in `readings` which (now at least) only get created if you have specified that readings should be stored locally rather than uploaded.

maddogeco, 2022-11-15:
[image]
yes it is the readings folder that is filling up. i have set it up to post via mqtt. should they be removed after the post happens ?

MrDrem, 2022-11-15:
I /think/ that this was the behaviour under version 0.0.2.

I have a feeling that I also managed to get it happening when I was trying to upload to two endpoints, which was almost certainly down to me bodging some bit of code badly.

ZodiusInfuser, 2022-11-17:
Being behaviour under v0.0.2 sounds plausible. @maddogeco I would suggest you try updating the code with what is in this branch: https://github.com/pimoroni/enviro/tree/patch

I know for sure that does not create readings files unless you explicitly say you want local logging.

ZodiusInfuser, 2022-11-26:
There is a new release that should fix this issue, so I'm closing this: https://github.com/pimoroni/enviro/releases/tag/v0.0.9
Please re-open it or raise a new issue if you update and experience the problem again. Thanks

closed 2022-11-26 by ZodiusInfuser