pimoroni/enviro #110: Make ntp server configurable during provisioning

peter-mount, 2022-11-02:
Currently phew is hardcoded to `pool.ntp.org` but for some environments this might not be accessible or desirable.

Being able to change this to another ntp server, in my instance I have my own stratum 1 ntp server locally, would mean that any clock synchronization would not rely on a potentially unstable internet connection.

I've raised pimoroni/phew#16 which covers the part in phew where this would need to be done to support this during enviro's provisioning.

ZodiusInfuser, 2022-11-04:
Thanks for suggesting this. As the firmware is just local to your PicoW you can edit the relevant code line on your specific board if you wish to get something working now.

I do agree though that being able to specify it would be preferable. Were you thinking it would be something the user gets asked during provisioning? I can imagine most users don't know what ntp loop to choose, so I'm not sure about it being an always present setup step. Perhaps just a line in the config file that you can edit?

I'm curious what your thoughts are

peter-mount, 2022-11-04:
I'm doing it currently by manually editing the file.

I raised this because not everyone would know where to do this. If it's a config entry then that would be good as an alternative to provisioning.

MrDrem, 2022-11-04:
I think that this might be resolved by better documentation on where things like this are within the code, rather than spraying too many options around.

Anyone who wants to edit the ntp server address is likely to know what one looks like, so can find it be searching, but it being flagged either with a comment or in general documentation (if space saving is a issue) would be useful.