pimoroni/enviro #112: Enviro_Weather hangs - Random, no indication as to why.

dave-ct, 2022-11-08:
Has happened a couple of times now where the board just hangs after a scheduled wakeup.  Not sure why as has gone days before without any issue.  Does not seem related to network as doe snot even attempt to connect to wifi, just the white LED pulses.  

Logs are below and can see from  the wakup at 19:00 (marked with stars) it just hangs and then next log entry is when I removed power at 19:39 and plugged it in again.  No issue with battery as still at 3.9v.   Its set to take readings evey 10 minutes and using local MQTT for destination. 

```
2022-11-08 18:50:02 [debug    / 118kB] > performing startup
2022-11-08 18:50:02 [info     / 124kB]   - wake reason: unknown
2022-11-08 18:50:02 [debug    / 122kB]   - turn on activity led
2022-11-08 18:50:03 [info     / 111kB] > 4 cache files need uploading
2022-11-08 18:50:03 [info     / 108kB] > connecting to wifi network 'VM2878914-5G_EXT'
2022-11-08 18:50:05 [debug    / 102kB]   - connecting
2022-11-08 18:50:08 [debug    /  89kB]   - connecting
2022-11-08 18:50:09 [info     /  86kB]   - ip address:  192.168.0.90
2022-11-08 18:50:10 [info     / 109kB]   - uploaded 2022-11-08T18:20:03Z.json to mqtt
2022-11-08 18:50:10 [info     / 102kB]   - uploaded 2022-11-08T18:30:03Z.json to mqtt
2022-11-08 18:50:11 [info     /  96kB]   - uploaded 2022-11-08T18:40:03Z.json to mqtt
2022-11-08 18:50:11 [info     / 117kB]   - uploaded 2022-11-08T18:50:03Z.json to mqtt
2022-11-08 18:50:11 [info     / 115kB] > going to sleep
2022-11-08 18:50:11 [debug    / 113kB]   - clearing and disabling timer and alarm
2022-11-08 18:50:12 [info     / 111kB]   - setting alarm to wake at 19:00pm
2022-11-08 18:50:12 [info     / 109kB]   - shutting down
**2022-11-08 19:00:02 [debug    / 118kB] > performing startup
**2022-11-08 19:00:02 [info     / 124kB]   - wake reason: unknown
**2022-11-08 19:00:02 [debug    / 122kB]   - turn on activity led
2022-11-08 19:39:09 [debug    / 118kB] > performing startup
2022-11-08 19:39:09 [info     / 124kB]   - wake reason: unknown
2022-11-08 19:39:09 [debug    / 122kB]   - turn on activity led
2022-11-08 19:39:10 [info     / 111kB] > going to sleep
2022-11-08 19:39:10 [debug    / 109kB]   - clearing and disabling timer and alarm
2022-11-08 19:39:10 [info     / 107kB]   - setting alarm to wake at 19:40pm
2022-11-08 19:39:10 [info     / 105kB]   - shutting down
2022-11-08 19:39:11 [debug    /  88kB]   - on usb power (so can't shutdown) halt and reset instead
```

ZodiusInfuser, 2022-11-15:
Hi @dave-ct. Thanks for raising this. I suspect there is some internal error in the code that's not being logged. To help diagnose this, I have pushed change https://github.com/pimoroni/enviro/pull/113/commits/9d41d2d869d3711dc6ff0125f0e9103b40b4286b to the `patch` branch that captures any exceptions and logs them. If you are able to re-run with that change added (or all of the fixes from the `patch` branch) that would be appreciated.

dave-ct, 2022-11-18:
Thanks, I have deployed the patch branch and will monitor to see it it happens with this version.  FYI the rain issue #70 looks to be fixed and working well now.

dave-ct, 2022-11-22:
@ZodiusInfuser  Have had the patch verison running over the last 4 days and the issue has not happened again so suspect something in this has fixed it.  Will close this issue.

closed 2022-11-22 by dave-ct

ZodiusInfuser, 2022-11-22:
Thank you @dave-ct. The pessimist in my isn't entirely convinced it's fixed, but the last changes I did do let the board handle errors a bit more gracefully. If it comes up again, or you see anything interesting in the log, feel free to re-open this issue.