pimoroni/enviro #116: Using sensors via Qw/ST connector

kscarlett, 2022-11-18:
I've tried to get my SCD41 (CO2) sensor to work via the Qw/ST connector, but can't seem to get it to work at all.

Are there any plans for either simplifying the process, or adding documentation to cover this topic?

ZodiusInfuser, 2022-11-18:
Are you able to post the code you tried? The process should just involve you modifying main.py to include the setup for the sensor, taking of the reading, and adding of the reading to `readings`. I confess though that I have not attempted this myself yet.

I don't have the sensor to hand, but if you're happy to experiment, then here is my best guess at how `main.py` should be modified:
```python
# import enviro firmware, this will trigger provisioning if needed
import enviro
import breakout_scd41

# initialise enviro
enviro.startup()

# now that we know the device is provisioned import the config
try:
  import config
except:
  enviro.halt("! failed to load config.py")

# if the clock isn't set...
if not enviro.is_clock_set():
  enviro.logging.info("> clock not set, synchronise from ntp server")
  if not enviro.sync_clock_from_ntp():
    # failed to talk to ntp server go back to sleep for another cycle
    enviro.halt("! failed to synchronise clock")
  enviro.logging.info("  - rtc synched")      

# check disk space...
if enviro.low_disk_space():
  # less than 10% of diskspace left, this probably means cached results
  # are not getting uploaded so warn the user and halt with an error
  enviro.halt("! low disk space")

# take a reading from the onboard sensors
reading = enviro.get_sensor_readings()

# here you can customise the sensor readings by adding extra information
# or removing readings that you don't want, for example:
# 
#   del readings["temperature"]        # remove the temperature reading
#
#   readings["custom"] = my_reading()  # add my custom reading value

breakout_scd41.init(enviro.i2c)
breakout_scd41.start()

while not breakout_scd41.ready():
    pass

co2, temperature, humidity = breakout_scd41.measure()
readings["scd_co2"] = co2
readings["scd_temp"] = temperature
readings["scd_humid"] = humidity

# is an upload destination set?
if enviro.config.destination:
  # if so cache this reading for upload later
  enviro.cache_upload(reading)

  # if we have enough cached uploads...
  if enviro.is_upload_needed():
    enviro.logging.info(f"> {enviro.cached_upload_count()} cache files need uploading")
    if not enviro.upload_readings():
      enviro.halt("! reading upload failed")
else:
  # otherwise save reading to local csv file (look in "/readings")
  enviro.save_reading(reading)

# go to sleep until our next scheduled reading
enviro.sleep()
```
Let me know if this works, or what errors it comes back with.

ZodiusInfuser, 2022-11-22:
Glad this worked for you, with edits 😅 . I agree we should add documentation for this, so I will leave this issue open as a reminder.

sjefferson99, 2023-07-08:
@ZodiusInfuser I've fixed the comment typo and added some documentation linked from the code comment to make this a little easier for people to add custom data points. In PR #184