pimoroni/enviro #117: Urban board won't boot with v0.0.9

peter-mount, 2022-11-26:
I upgraded both of my Enviro Weather and Urban boards to 0.0.9 last night and although the Weather board worked fine, the Urban board wouldn't boot whilst connected to USB after it was reprovisioned.

In dmesg all I could see was
```
[ 3838.099979] usb usb3-port3: attempt power cycle
[ 3838.511780] usb 3-3: new full-speed USB device number 104 using xhci_hcd
[ 3838.512863] usb 3-3: Device not responding to setup address.
[ 3838.720883] usb 3-3: Device not responding to setup address.
[ 3838.927769] usb 3-3: device not accepting address 104, error -71
```
I could only get it to respond if I held the poke button on restart (but not long enough to enter provisioning mode).

After some hunting around I found that this is a known issue to happen at times with MicroPython, e.g. micropython/micropython#9605

Do by adding the following at the start of main.py seems to have solved the problem:
```
from time import sleep
sleep(1)
```

ZodiusInfuser, 2022-11-26:
Thank you for raising this. Curious why it only happens with your Enviro Urban. Perhaps that's just coincidence.
A sleep would be easy enough to add, although I wonder if it could be for a shorter time, like 0.1s?

peter-mount, 2022-11-29:
I've just tried it and the shortest that works is 0.5s.

When I tried 0.1 or 0.2 then the USB stack didn't start and had to hold poke down to get it to boot.