pimoroni/enviro #118: WiFi password truncated by provisioning 0.0.9

TechWilk, 2022-11-26:
Board: Enviro Indoor

Having:

- flashed `0.0.9`
- connected to the pico access point
- followed the provisioning steps, including selecting a wifi network and entering its password
- finished provisioning
- disconnected from computer
- connected battery pack
- turned on batter pack

The enviro board fails to connect to the wifi network, with a pulsing red warning led.
Note that the WiFi password does not contain single quotes, but has a combination of letters numbers and symbols

Workaround:

- Connect the pico to thonny
- load the config.py file
- notice that only the first 15 chars of the wifi password is saved
- enter the remaining wifi password
- save and reboot
- this resolves the problem

(I have not tested previous firmware, since I have only just got the board)

ZodiusInfuser, 2022-11-26:
Thanks for raising this. I will attempt to reproduce this next week, but from a glance at the code now there is no obvious limit on the length of the password supported. Relevant files are:
https://github.com/pimoroni/enviro/blob/d091f7539acc9513c391f0eec498c0cc898edb7a/enviro/provisioning.py#L84
and
https://github.com/pimoroni/enviro/blob/d091f7539acc9513c391f0eec498c0cc898edb7a/enviro/html/provision-step-2-wifi.html#L29
I am glad that editing the config.py lets you get around this

MrDrem, 2022-11-27:
My WiFi password is 16 chars, and checking it on my newly upgraded Urban board, it isn't truncated.

TechWilk, 2022-11-30:
Yes it seemed rather odd, perhaps that it got stuck on a special character rather than a character limit as such?
The password in question is much longer and contains a lot of special chars. I'll try again in the next few days when another board arrives and see if I can reproduce it reliably

ZodiusInfuser, 2022-12-01:
Truncating on a special character could be completely possible, though I do not know where such a check would be happening within the code yet.

basil-dsouza, 2022-12-02:
I have faced the same issues in version 0.0.8 as well. 
What I can remember doing is:
* Flash board from stock to 0.0.8
* Manually modify config.py to set the wifi, etc
* However it then didnt connect to wifi
* So I set Provisioned to false and then reset it
* Connected to self wifi and did the provisioning via that approach
* It auto filled all of the values, but when I completed and connected back to usb i noticed that the wifi password was truncated (didnt count the exact number of letters)
* I can confirm that I did not have any special characters in the password

Lipod, 2023-08-19:
I am currently setting up my enviro urban and observed this issue with the stock firmware. I have now flashed it with 0.0.9 and encountered the issue again when I used the web page to provision after the upgrade. My WiFi key is 22 character alphanumeric (no special characters) and I am certain that I type the correct key into the web page, but when I opened the config file using Thonny the last character had been chopped.

Before I tried Thonny it was frustrating at to keep re-entering provisioning mode and finding the wrong key in the text box.

Jessepo, 2023-11-02:
Hi, I had a similar issue when the provisioning set my all numeric password as a numeric instead of a string. I then saw errors thrown when running main.py in thonny, went to the config.py and added "" around the password and it worked. Should I create an issue?