pimoroni/enviro #120: [enviro][0.0.9][grow] - Pico W refuses to Connect to WiFi

byronczimmer, 2022-11-27:
Loaded pimoroni-picow_enviro-v1.19.10-micropython-v0.0.9.uf2 onto the enviro-grow that I've been trying to get working for the past few days.

I am getting repeated failures connecting to my local WiFi.

When it does connect to the WiFi, adafruit.io will of course limit the number of updates it will accept before throwing an error, making the grow hold off for a minute ( which I've changed to 2 minutes to prevent adafruit from throwing a full throttle ).  However, sometimes when it comes back - it's not reconnecting and the warning light just happily flashes away.

I have managed to get the chip to talk to adafruit.io a few times.
1) It will connect after the RP2040 chip has been OFF for quite a while.
2) It will connect if I've previously run a separate 'test' program for as long as it took to connect, and then started main.py from Thonny.

This is an intermittent issue and is proving difficult to track down.
I suspect there is some issue with the wifi chip independently maintaining charge from the main RP2040 chip.

ZodiusInfuser, 2022-11-27:
I too have been experiencing intermittent WiFi connection issues, and they are indeed frustrating to track down.
```
I suspect there is some issue with the wifi chip independently maintaining charge from the main RP2040 chip.
```
This is quite likely when running off USB, as there is no direct way for the code to force a reset of the WiFi. If you're running off battery, then it will reset as the entire 3.3V rail gets turned off to enter deep sleep.

Then there's whatever your WiFi router is doing. I've had at least one occasion where connections kept failing until I power cycled my router. Perhaps the router had put the Enviro board on a temporary block list??

byronczimmer, 2022-12-06:
I have found elsewhere that setting the powersaving mode OFF:
wlan.config(pm = 0xa11140)
might help.

And on a fresh Pico-W ( newly soldered this evening ) it did.

I'll test that on the envrio-grow soon enough, but figured I'd post here.

( yes, I know the goal is low power overall, but if the device isn't connecting and offloading data -- it doesn't matter how much power is saved )

jxmesw1, 2023-01-01:
I have been experiencing this too. I have an Enviro Weather powered via USB and every so often it refuses to connect to the network. Power cycling the board seems to work, but resetting it via the button does not. I have tried to replicate the issue when the board is powered by a battery pack however I have not been able to do so. 

Like @byronczimmer said, I believe it is something to do with the WiFi chip maintaining charge.  I will experiment tomorrow by modifying the firmware and will post my results here.

jxmesw1, 2023-01-03:
Update - I've not been able to reproduce the error while the board is on battery.