pimoroni/enviro #124: NTP / timestamps suddenly wrong

joshkopecek, 2022-12-05:
I just managed to get the enviro working, but in the last day the timestamps jumped. I think this may have been since I upgraded to v0.0.9 of the firmware? It now seems to think it's Christmas day…

- I'm running the [Enviro Weather ](https://shop.pimoroni.com/products/enviro-weather-board-only?variant=40056047173715)
- I upgraded to [v0.0.9](https://github.com/pimoroni/enviro/releases/tag/v0.0.9) firmware (the link in the release description to the uf2 file returns 404, so I used the '...filesystem-only.uf2' one)
- I tried reporting direct to Adafruit.io, but this started jumping to the 24th December. 
- The exact difference between now and the reported timestamp is 19 days, 12 hours and 22 minutes.
- I tried reporting to a custom http endpoint locally, and exactly the same happens.
- It's connecting to WiFi fine and reporting via the internet to Adafruit fine, so connection isn't an issue.

Should I post the log?

[image]

ZodiusInfuser, 2022-12-05:
Thanks for the note about the link being wrong on the release. I have now corrected this

joshkopecek, 2022-12-05:
@ZodiusInfuser thanks for the tip. I'll try to reset the time, but I have no idea how it could have got offset like that. Will report further.

joshkopecek, 2022-12-05:
That did it – powering off for > 1 min. When I upgraded to v0.0.9, that's when the offset occurred (and stuck). I may not have completely powered it off for more than a minute.