pimoroni/enviro #126: Enviro (urban/weather/indoor) - Low Disk Space after Wifi Outage - Edge case

dave-ct, 2022-12-07:
Version: 0.0.9
Image: https://github.com/pimoroni/enviro/actions/runs/3624876714
Summary: Uploads folder fills up and cannot recover without deleteing files.
Suggested Fix: Update so that if the file system is 90% full then do not take further readings but attempt to do uploads only on each wake so that it can eventually recover instead of sleeping. 

While testing anothet issue had 2 x Urbans, 2 x Indoors and 1 x Weather running on battery doing readings every 2 minutes and uploads every 3 readings.  At about 08:20 the wifi router seems to have gone down so when inspecting the devices 12 hours later all flashing read.  After rebooting the wifi router the devices did not automtically recover.  Tried poking but reverted quickly back to red light.  On inspection of device logs it shows  low disk space.

```
2022-12-07 20:09:52 [debug    / 115kB] > performing startup
2022-12-07 20:09:52 [info     / 123kB]   - wake reason: button
2022-12-07 20:09:52 [debug    / 121kB]   - turn on activity led
2022-12-07 20:09:53 [error    / 119kB] ! low disk space
2022-12-07 20:09:53 [info     / 116kB] > going to sleep
2022-12-07 20:09:53 [debug    / 114kB]   - clearing and disabling previous alarm
2022-12-07 20:09:53 [info     / 112kB]   - setting alarm to wake at 20:10pm
2022-12-07 20:09:53 [info     / 110kB]   - shutting down
2022-12-07 20:10:03 [debug    / 115kB] > performing startup
2022-12-07 20:10:03 [info     / 123kB]   - wake reason: rtc_alarm
2022-12-07 20:10:03 [debug    / 121kB]   - turn on activity led
2022-12-07 20:10:04 [error    / 119kB] ! low disk space
2022-12-07 20:10:04 [info     / 116kB] > going to sleep
2022-12-07 20:10:04 [debug    / 114kB]   - clearing and disabling previous alarm
2022-12-07 20:10:04 [info     / 112kB]   - setting alarm to wake at 20:12pm
2022-12-07 20:10:04 [info     / 110kB]   - shutting down
2022-12-07 20:10:17 [debug    / 115kB] > performing startup
2022-12-07 20:10:17 [info     / 123kB]   - wake reason: button
2022-12-07 20:10:17 [debug    / 121kB]   - turn on activity led
2022-12-07 20:10:18 [error    / 119kB] ! low disk space
2022-12-07 20:10:18 [info     / 116kB] > going to sleep
2022-12-07 20:10:18 [debug    / 114kB]   - clearing and disabling previous alarm
2022-12-07 20:10:18 [info     / 112kB]   - setting alarm to wake at 20:12pm
2022-12-07 20:10:18 [info     / 110kB]   - shutting down
```

Uploads folder full of about 180 json files waiting to upload.  

Simple fix was to delete the uplaods folder.

While this is an edge case due to frquency of readings possible solution is instead of `enviro.halt` at 10% then can attepemt to upload the readings but dont take any new ones, can log a alternate mesasge as well.  This could be useful in particular for the Urban or Weather as if they are attached to a roof or hard to reach area means you have to bring inside to fix  which is a pain.

I think I could make pull request for this, just need to alter `main.py` but wanted to check if alternate approach should be done?

ZodiusInfuser, 2022-12-09:
Thanks for raising this edge case. Yes, the board should keep attempting to upload files in this instance.

Here's how I'd probably do it. I suspect there's some scenario I'm not thinking of with this though, not to mention the duplicated code.
```python
import enviro
import os

try:
  # initialise enviro
  enviro.startup()

  # if the clock isn't set...
  if not enviro.is_clock_set():
    enviro.logging.info("> clock not set, synchronise from ntp server")
    if not enviro.sync_clock_from_ntp():
      # failed to talk to ntp server go back to sleep for another cycle
      enviro.halt("! failed to synchronise clock")  

  # check disk space...
  if enviro.low_disk_space():
    # less than 10% of diskspace left, this probably means cached results
    # are not getting uploaded so warn the user and halt with an error

    # is an upload destination set?
    if enviro.config.destination:
      enviro.logging.error("! low disk space. Attempting to upload file(s)")

      # if we have enough cached uploads...
      enviro.logging.info(f"> {enviro.cached_upload_count()} cache file(s) need uploading")
      if not enviro.upload_readings():
        enviro.halt("! reading upload failed")
    else:
      # no destination so go to sleep
      enviro.halt("! low disk space")

  # TODO this seems to be useful to keep around?
  filesystem_stats = os.statvfs(".")
  enviro.logging.debug(f"> {filesystem_stats[3]} blocks free out of {filesystem_stats[2]}")

  # TODO should the board auto take a reading when the timer has been set, or wait for the time?
  # take a reading from the onboard sensors
  enviro.logging.debug(f"> taking new reading")
  reading = enviro.get_sensor_readings()

  # here you can customise the sensor readings by adding extra information
  # or removing readings that you don't want, for example:
  # 
  #   del readings["temperature"]        # remove the temperature reading
  #
  #   readings["custom"] = my_reading()  # add my custom reading value

  # is an upload destination set?
  if enviro.config.destination:
    # if so cache this reading for upload later
    enviro.logging.debug(f"> caching reading for upload")
    enviro.cache_upload(reading)

    # if we have enough cached uploads...
    if enviro.is_upload_needed():
      enviro.logging.info(f"> {enviro.cached_upload_count()} cache file(s) need uploading")
      if not enviro.upload_readings():
        enviro.halt("! reading upload failed")
    else:
      enviro.logging.info(f"> {enviro.cached_upload_count()} cache file(s) not being uploaded. Waiting until there are {enviro.config.upload_frequency} file(s)")
  else:
    # otherwise save reading to local csv file (look in "/readings")
    enviro.logging.debug(f"> saving reading locally")
    enviro.save_reading(reading)

  # go to sleep until our next scheduled reading
  enviro.sleep()

# handle any unexpected exception that has occurred
except Exception as exc:
  enviro.exception(exc)
```
If you could give it a test that would be great. And yes, if you find that it does work for you, or need to make modifications, please raise a PR.

dave-ct, 2022-12-11:
**Just stroiring LogFile during test - Will update once test complete**

Started testing by disabling wifi router and letting 2 x Urbans, 2 x Indoors and 1 x Weather build up files and fill up (readings evey 2 minutes with uplaods every 3 minutes).

Sample log before turning wfi router back on, see entries at 17:12 which show not taking readings an atemtping to upload (162 files)  but will fail as wifi still off. :
```
2022-12-11 17:10:04 [debug    / 115kB] > performing startup
2022-12-11 17:10:04 [info     / 122kB]   - wake reason: rtc_alarm
2022-12-11 17:10:04 [debug    / 120kB]   - turn on activity led
2022-12-11 17:10:04 [debug    / 118kB] > 23 blocks free out of 212
2022-12-11 17:10:05 [debug    / 115kB] > taking new reading
2022-12-11 17:10:05 [info     / 110kB]   - seconds since last reading: 120
2022-12-11 17:10:05 [debug    / 107kB]   - starting sensor
2022-12-11 17:10:05 [debug    / 105kB]   - wait 5 seconds for airflow
2022-12-11 17:10:10 [debug    /  86kB]   - taking pms5003i reading
2022-12-11 17:10:10 [debug    /  84kB]   - taking microphone reading
2022-12-11 17:10:11 [debug    / 116kB] > caching reading for upload
2022-12-11 17:10:11 [info     /  83kB] > 162 cache file(s) need uploading
2022-12-11 17:10:11 [info     /  80kB] > connecting to wifi network 'CT_752091'
2022-12-11 17:10:13 [error    / 118kB] ! failed to connect to wireless network CT_752091
2022-12-11 17:10:14 [error    / 115kB]   - cannot upload readings, wifi connection failed
2022-12-11 17:10:14 [error    / 113kB] ! reading upload failed
2022-12-11 17:10:14 [info     / 111kB] > going to sleep
2022-12-11 17:10:14 [debug    / 109kB]   - clearing and disabling previous alarm
2022-12-11 17:10:14 [info     / 107kB]   - setting alarm to wake at 17:12pm
2022-12-11 17:10:14 [info     / 105kB]   - shutting down
2022-12-11 17:12:04 [debug    / 115kB] > performing startup
2022-12-11 17:12:04 [info     / 122kB]   - wake reason: rtc_alarm
2022-12-11 17:12:04 [debug    / 120kB]   - turn on activity led
2022-12-11 17:12:04 [error    / 118kB] ! low disk space. Attempting to upload file(s)
2022-12-11 17:12:05 [info     / 101kB] > 162 cache file(s) need uploading
2022-12-11 17:12:05 [info     /  98kB] > connecting to wifi network '<Removed>'
2022-12-11 17:12:07 [error    /  90kB] ! failed to connect to wireless network <Removed>
2022-12-11 17:12:07 [error    /  87kB]   - cannot upload readings, wifi connection failed
2022-12-11 17:12:07 [error    /  85kB] ! reading upload failed
2022-12-11 17:12:07 [info     / 124kB] > going to sleep
2022-12-11 17:12:07 [debug    / 122kB]   - clearing and disabling previous alarm
2022-12-11 17:12:07 [info     / 120kB]   - setting alarm to wake at 17:14pm
2022-12-11 17:12:07 [info     / 118kB]   - shutting down
```

ZodiusInfuser, 2022-12-13:
Glad to see that the logging change worked and let the board recover! I'll look at getting the fix merged in.

dave-ct, 2022-12-13:
@ZodiusInfuser have done a seperate pull request here #128

closed 2022-12-18 by dave-ct