pimoroni/enviro #127: Sending files to Influxdb not all data are received in Database - Wrong dateTime used

niksdk, 2022-12-11:
I have a set-up where data are send to a Influx DB I am hosting on my own Raspberry.

When several files are uploadet it look like not all files are updated into Influxdb.
In my case the enviro log tell that the files are uploaded 

 2022-12-11 16:25:34 [info     / 112kB]   - uploaded 2022-12-11T16:09:55Z.json to influxdb
2022-12-11 16:25:35 [info     /  97kB]   - uploaded 2022-12-11T16:12:32Z.json to influxdb
2022-12-11 16:25:35 [info     /  82kB]   - uploaded 2022-12-11T16:14:57Z.json to influxdb
2022-12-11 16:25:36 [info     / 110kB]   - uploaded 2022-12-11T16:20:01Z.json to influxdb
2022-12-11 16:25:36 [info     /  95kB]   - uploaded 2022-12-11T16:25:28Z.json to influxdb

But when I make a select in the database I only see 3 inserts. When I convert the time for the 3 rows, I can see that the time is when data is updated not when messured. 

time                device value
----                ------ -----
(11-12-22 16:25:34) 1670775934000000000 enviro 140.22
(11-12-22 16:25:35) 1670775935000000000 enviro 44.64
(11-12-22 16:25:36) 1670775936000000000 enviro 49.48

dave-ct, 2022-12-11:
@niksdk  Are you using 0.0.9 version?  As the code in their for influxdb desitination seems to take the timestamp from the readings as opposed to the time of upload which should give you the correct result.

niksdk, 2022-12-12:
Thanks, I have now updated to latest version and the time is correct in the Database.
But I see other problems, which I will create another issue for.

closed 2022-12-12 by niksdk