pimoroni/enviro #133: Enviro Weather resets when on USB power

guruthree, 2022-12-30:
I had my Enviro Weather connected over USB to Thonny to monitor debug output. Per the [store page](https://shop.pimoroni.com/products/enviro-weather?variant=40056776917075) "the board will never sleep if connected to USB power" and thus I expected to see the equivalent of the log.txt output. However, I instead observed Thonny's back-end crashing after loosing connection to the Pico W after a reset:
```
>>> import main
       ___            ___            ___          ___          ___            ___       
      /  /\          /__/\          /__/\        /  /\        /  /\          /  /\      
     /  /:/_         \  \:\         \  \:\      /  /:/       /  /::\        /  /::\     
    /  /:/ /\         \  \:\         \  \:\    /  /:/       /  /:/\:\      /  /:/\:\    
   /  /:/ /:/_    _____\__\:\    ___  \  \:\  /__/::\      /  /:/~/:/     /  /:/  \:\   
  /__/:/ /:/ /\  /__/::::::::\  /___\  \__\:\ \__\/\:\__  /__/:/ /:/___  /__/:/ \__\:\  
  \  \:\/:/ /:/  \  \:\~~~__\/  \  \:\ |  |:|    \  \:\/\ \  \:\/:::::/  \  \:\ /  /:/  
   \  \::/ /:/    \  \:\         \  \:\|  |:|     \__\::/  \  \::/~~~`    \  \:\  /:/   
    \  \:\/:/      \  \:\         \  \:\__|:|     /  /:/    \  \:\         \  \:\/:/    
     \  \::/        \  \:\         \  \::::/     /__/:/      \  \:\         \  \::/     
      \__\/          \__\/          `~~~~~`      \__\/        \__\/          \__\/      

    -  --  ---- -----=--==--===  hey enviro, let's go!  ===--==--=----- ----  --  -     

2000-01-01 00:00:19 [debug    / 108kB] > performing startup
2000-01-01 00:00:19 [info     / 115kB]   - wake reason: usb_powered
2000-01-01 00:00:19 [debug    / 113kB]   - turn on activity led
2000-01-01 00:00:19 [info     / 111kB] > clock not set, synchronise from ntp server
2000-01-01 00:00:19 [info     / 109kB] > connecting to wifi network 'gnet6'
2000-01-01 00:00:20 [info     /  91kB]   - ip address:  192.168.0.21
2022-12-30 14:04:19 [info     /  89kB]   - rtc synched
2022-12-30 14:04:19 [debug    / 118kB] > 83 blocks free out of 212
2022-12-30 14:04:19 [debug    / 115kB] > taking new reading
2022-12-30 14:04:19 [info     / 111kB]   - seconds since last reading: 1216
2022-12-30 14:04:20 [debug    / 102kB] > saving reading locally
2022-12-30 14:04:20 [info     /  97kB] > going to sleep
2022-12-30 14:04:20 [debug    /  95kB]   - clearing and disabling previous alarm
2022-12-30 14:04:20 [info     /  92kB]   - setting alarm to wake at 14:05pm
2022-12-30 14:04:21 [info     /  90kB]   - shutting down
2022-12-30 14:04:21 [debug    /  88kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2022-12-30 14:05:00 [debug    / 118kB]   - reset
PROBLEM IN THONNY'S BACK-END: Exception while handling 'execute_source' (ConnectionError: EOF).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.
```

This looks like a case of the product page being technically correct, but not explicit. If the board never sleeps, it also does not appear to operate uninterrupted. I was hoping it might be possible implement a background web server to query the board directly, but with constant resets it doesn't look like this would be easily doable.

I suggest:
* Modifying the product page to remove the quote about not sleeping on USB power as, while it may not enter low power mode, functionally it behaves identically to battery power and still calls `sleep()`
* Modifying [main/documentation/developer-guide.md](https://github.com/pimoroni/enviro/blob/main/documentation/developer-guide.md) to include the reset
* Adding a link to https://github.com/pimoroni/enviro/blob/main/documentation/developer-guide.md from the documentation section of https://github.com/pimoroni/enviro/blob/main/README.md

I think I might be able to create a PR for the last two if that would help.

jxmesw1, 2023-01-03:
I could be wrong, but I believe the reset is caused by the RTC alarming: it resets the board so the program can run from the very beginning. As with the sleeping on USB, I think the board only realises it is on usb power when it fails to go to sleep.

I've not had a proper look at the firmware yet, so please correct me if I'm wrong.

ZodiusInfuser, 2023-01-05:
@guruthree Yes, this is indeed a case of us being technically correct, as when on USB power the board cannot physically enter the "deep sleep", but that says nothing of the board resetting. I'll get Hel to update our product pages with a note about the reset on USB.

helgibbons, 2023-01-05:
Technically correct, the best kind of correct!

I've updated the shop pages :)

closed 2023-01-05 by ZodiusInfuser