pimoroni/enviro #138: enviro indoor picow 0.0.9 jittery temperature

herbsoft, 2023-01-03:
As shown in the image I'm getting jittery results from the picow indoor around temperature readings

[image]

Looks to be a variation of around 0.8C on readings taken every 15 mins

herbsoft, 2023-01-03:
_(That +3C in the name is just to remind me that it's reporting about 3C above the actual temperature)._

ZodiusInfuser, 2023-01-05:
Hi. That is unusual. May I ask, how frequently are you uploading your readings? And how are you powering your board?

herbsoft, 2023-01-16:
Sorry taken a while to get back to this, I was uploading every 4 results and powering over USB.

MrDrem, 2023-01-24:
I see this two of my three devices, the Enviro Urban and Weather. It does occur in the Indoor, but it's not as obvious, I assume that this is because it doesn't have as much additional hardware on it.

herbsoft, 2023-07-30:
Has there been any improvement here? If not I'm wondering about trying to alter the code to do the data upload on a different cycle to reading the data.