pimoroni/enviro #139: No module named 'wakeup' in v0.0.9 adafruit_io_fixes PR

djbloc, 2023-01-04:
Nuke flashed Pico and flashed using pimoroni-picow-v1.19.10-micropython.uf2 and manually uploaded the latest code from https://github.com/pimoroni/enviro/tree/patch/adafruit_io_fixes/enviro.

In log.txt it states:

```
2000-01-01 01:10:05 [info     / 131kB] > performing startup
2000-01-01 01:10:05 [debug    / 129kB]   - running Enviro 0.0.9, MicroPython 9dfabcd on 2022-11-18
2000-01-01 01:10:05 [exception / 126kB] ! Traceback (most recent call last):
  File "main.py", line 37, in <module>
  File "enviro/__init__.py", line 474, in startup
  File "enviro/__init__.py", line 305, in get_wake_reason
ImportError: no module named 'wakeup'

2000-01-01 01:10:05 [info     / 123kB] > going to sleep
2000-01-01 01:10:05 [debug    / 122kB]   - clearing and disabling previous alarm
2000-01-01 01:10:05 [info     / 120kB]   - setting alarm to wake at 01:15am
2000-01-01 01:10:05 [info     / 118kB]   - shutting down
2000-01-01 01:10:05 [debug    / 136kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
```

Can't find wakeup.py searching Github in main branch or PR. Code was added in PR https://github.com/pimoroni/enviro/pull/113/files. Is there a file missing?

dave-ct, 2023-01-04:
You could try installing
[ ](https://github.com/pimoroni/enviro/releases/download/v0.0.9/pimoroni-picow_enviro-v1.19.10-micropython-v0.0.9.uf2)
Then just copy over the upodated files from the patch branch to see if that works?

ZodiusInfuser, 2023-01-05:
Hi @djbloc. Wakeup is a module over in Pimoroni Pico (https://github.com/pimoroni/pimoroni-pico/tree/main/micropython/modules/wakeup), so you need to flash your Enviro with a recent version of the pimoroni-picow_enviro Micropython, like the one dave-ct linked to.

djbloc, 2023-01-05:
Thanks @dave-ct, @ZodiusInfuser. Will try that.

closed 2023-01-05 by djbloc