pimoroni/enviro #148: enviro indoor pico

foxalite, 2023-02-12:
1: With original firmware (0.0.2) I'm unable to connect to to pico by wifi, seem the dns server don't work.

2: Upgrade to firmware rev (0.0.9) I can setup the device (provisioning). After reset the device doe's nothing, no light etc. and I can't connect with thonny, so no log available.

3: Fresh flash with 0.0.9 without provisioning, I can connect to the device and setup de config.py. After save and restart de device, still doe's nothing, and still unable to connect with thonny.

4: Repeat step 3 without restart the device, and run main.py, device send 1 (one) time the  datas to the mqtt broquer, and nothing more.

What's wrong ?

ZodiusInfuser, 2023-02-12:
Hi @foxalite! That's a new one to me... Are you able to access the files on the Pico and post what the log.txt says?

foxalite, 2023-02-12:
> Hi @foxalite! That's a new one to me... Are you able to access the files on the Pico and post what the log.txt says?

No, I'm unable to connect to the pico with thony (see point 3)
I can connect only without provisioning or with filesystem only .

Other thing I do, just try to blink onboard led with thony (with gpio 0) without succes (btw this don't work with a single pico w board also). But this work with arduino, so I think the pico is OK.

ZodiusInfuser, 2023-02-12:
The LED on Pico W is not connected to GPIO0, but rather attached to the WiFi chip. It can be tested with this code:
```python
import time
import machine
led = machine.Pin("LED", machine.Pin.OUT)

while True:
    led.off()
    time.sleep(1)
    led.on()
    time.sleep(1)
```

Could you try flashing the board with this version of micropython? https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.10/pimoroni-picow_enviro-v1.19.10-micropython.uf2

It's the same version as used by Enviro, but without the file system. You should try and nuke the board before doing this though: https://datasheets.raspberrypi.com/soft/flash_nuke.uf2

foxalite, 2023-02-12:
Hi ZodiuInfuser,
Same issue than v1.19.12

And thank you for the pin, seems the official documentation is wrong [https://datasheets.raspberrypi.com/picow/PicoW-A4-Pinout.pdf](https://datasheets.raspberrypi.com/picow/PicoW-A4-Pinout.pdf)

ZodiusInfuser, 2023-02-12:
Sorry, could you elaborate on "same issue than v1.19.12"?
Are you saying you're not able to interact with it even after flash_nuke and putting v1.19.10 on? Have you unplugged power after flashing? If so then I'm thinking that the board may be faulty, so it could be worth messaging support@pimoroni.com to ask for a replacement.

That official pinout is correct, they've just chosen a colour that's hard to read. It actually says WL_GPIO0, which is GPIO0 on the WiFi chip.

foxalite, 2023-02-13:
Sorry for the misunderstanding, I was talking about this version [pimoroni-picow_enviro-v1.19.12-micropython.uf2](https://github.com/pimoroni/pimoroni-pico/releases/download/v1.19.12/pimoroni-picow_enviro-v1.19.12-micropython.uf2). that I also tried.
Yes, I have unplugged the board. 

I am able to access the board in all case before provisioning, but not after.
Another thing after power on the LED just light 1 or 2 sec in yellow and nothing more.

Anyway Thanks a lot for your help and your time.

foxalite, 2023-02-13:
Problem solved.
This append when the module is powered by usb on my computer port.
The module never wake up and I'm not able to connect with thony because:
dmesg:
[39251.199179] usb 1-2: new full-speed USB device number 121 using xhci_hcd
[39251.199494] usb 1-2: Device not responding to setup address.
[39251.407727] usb 1-2: Device not responding to setup address.
[39251.615168] usb 1-2: device not accepting address 121, error -71
[39251.615327] usb usb1-port2: unable to enumerate USB device

So no more /dev/ttyACM0.