pimoroni/enviro #152: Error uploading to adafruit.io

ittybittyspider, 2023-02-15:
I set up my Envirogrow, but it won't upload to adafruit.io. I have updated to the newest firmware. At first it just wouldn't upload, and I wouldn't see any errors in the adafruit.io monitor:

2023-02-15 15:35:05 [info     / 107kB] > uploading cached readings to Adafruit.io: plantynanny
2023-02-15 15:35:06 [debug    /  93kB]   - upload issue (404 b'Not Found')
2023-02-15 15:35:06 [error    /  91kB]   ! failed to upload '2023-02-15T15_16_45Z.json' to adafruit_io
2023-02-15 15:35:06 [error    /  89kB] ! reading upload failed
2023-02-15 15:35:06 [info     /  87kB] > going to sleep
2023-02-15 15:35:06 [debug    /  85kB]   - clearing and disabling previous alarm
2023-02-15 15:35:06 [info     / 120kB]   - setting alarm to wake at 15:45pm
2023-02-15 15:35:06 [info     / 118kB]   - shutting down
2023-02-15 15:35:06 [debug    / 116kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

I took a look in the adafruit_io.py file and the URL seemed to be wrong, so I corrected it from:

url = f"http://io.adafruit.com/api/v2/{username}/groups/enviro/data"

to

url = f"http://io.adafruit.com/api/v2/{username}/feeds/enviro/data"

Upload still failed, but I get the following error in the adafruit.io monitor:
MQTT ERROR: error saving data to plantynanny/feeds/enviro, data missing required value

adafruit_io username and key are correct.

Here is an example of one of the json files:
{"nickname": "gardener", "uid": "e6614103e744a522", "timestamp": "2023-02-15T15:39:44Z", "readings": {"temperature": 26.59, "humidity": 33.17, "pressure": 1017.4, "luminance": 0.0, "moisture_a": 0.0, "moisture_b": 0.0, "moisture_c": 0.0, "voltage": 0.0}, "model": "grow"}

Any ideas what is wrong?

helgibbons, 2023-02-15:
Have you created a group called 'enviro' on adafruit.io?

It's the button on the right (easy to mis-click and create a new feed instead!)

[image]

ittybittyspider, 2023-02-15:
Wow. Doh! 🤦🏼‍♂️ Thank you so much @helgibbons 🌟 It works now.

closed 2023-02-17 by ZodiusInfuser