pimoroni/enviro #154: Add Home Assistant Discovery

andymanic, 2023-02-24:
I'd love if these boards could actually be discovered by Home Assistant so it can add them as an MQTT device. As it stands you have to edit your configuration.yaml file to manually add entities, but other devices offer HASS discovery. This post [here](https://community.home-assistant.io/t/mqtt-auto-discovery-and-json-payload/409459/4) details the discovery MQTT command nicely - and I think just an extra toggle in the MQTT settings menu for "Home Assistant Discovery" would be perfect. 

I don't have much experience in python or with RP2040's yet, so I'm not sure I'm qualified to write this feature myself, but if I get the chance I'll certainly give it a go - but anyone more qualified is more than welcome to do it better than I can anyway!

andymanic, 2023-02-25:
While I remember, here's the full config for manually adding the enviro weather board to HASS (using Mosquitto). Add this to your configuration.yaml file then restart HASS. I've left the battery voltage option for when that's returned. It only adds the entities though, rather than a full device. 

It's also worth noting that Home Assistant doesn't seem to be able to support multiple data packets at once based on their timestamp. It doesn't matter much for me since mine only seems to be outputting a single result anyway despite being set to take new readings every 15 minutes and publishing after 10 readings, but it would be nice to have more frequent updates AND support for that data to be mapped correctly in Home Assistant. 

`mqtt:
    sensor:
    # Temperature
      - name: "Outdoor Temperature (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.temperature }}"
        unit_of_measurement: "°C"
        state_class: "measurement"
        device_class: "temperature"
        unique_id: "sensor.wind-station.temp"
    # Humidity
      - name: "Outdoor Humidity (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.humidity }}"
        unit_of_measurement: "%"
        state_class: "measurement"
        device_class: "humidity"
        unique_id: "sensor.wind-station.humidity"
    # Pressure
      - name: "Outdoor Pressure (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.pressure }}"
        unit_of_measurement: "mbar"
        state_class: "measurement"
        device_class: "pressure"
        unique_id: "sensor.wind-station.pressure"
    # Voltage
      - name: "Battery voltage (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.voltage }}"
        unit_of_measurement: "V"
        state_class: "measurement"
        device_class: "voltage"
        unique_id: "sensor.wind-station.voltage"
    # Rain
      - name: "Rain (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.rain }}"
        unit_of_measurement: "mm"
        state_class: "measurement"
        device_class: "precipitation"
        unique_id: "sensor.wind-station.rain"
    # Wind Speed
      - name: "Wind Speed (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.wind_speed }}"
        unit_of_measurement: "m/s"
        state_class: "measurement"
        device_class: "wind_speed"
        unique_id: "sensor.wind-station.wind_speed"
    # Wind Direcction
      - name: "Wind Direction (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.wind_direction }}"
        unit_of_measurement: "°"
        state_class: "measurement"
        unique_id: "sensor.wind-station.wind_direction"
    # Light
      - name: "Outdoor Light Level (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.light }}"
        unit_of_measurement: "lx"
        state_class: "measurement"
        device_class: "illuminance"
        unique_id: "sensor.wind-station.light"
`

helgibbons, 2023-02-27:
I ran into the same issue when posting my data into Home Assistant - as far as I can tell it has no concept of historical data, so it just associates all the data with the time it is posted in. To get round this, I ended up getting my Enviros to post into InfluxDB and hooked my Influx DB instance up to Home Assistant/Grafana.

I've not had much luck getting MQTT auto-discovery working generally with Home Assistant - if someone does work out the magic incantations for Enviro I'd be super interested to know them :)

andymanic, 2023-02-27:
> I ran into the same issue when posting my data into Home Assistant - as far as I can tell it has no concept of historical data, so it just associates all the data with the time it is posted in. To [...]

Yeah - Influx is a must-have with HASS, and happily the InfluxDB addon is really simple to set up and get working! Still, as far as I can tell it still uses time-of-arrival rather than the timestamp included (at least via MQTT).

While I'm here again, since I've had to update to V0.0.9 as the shipping version flat out stopped working for me, I've updated my HASS config to support the new format: 
`mqtt:
    sensor:
    # Temperature
      - name: "Outdoor Temperature (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.temperature }}"
        unit_of_measurement: "°C"
        state_class: "measurement"
        device_class: "temperature"
        unique_id: "sensor.wind-station.temp"
    # Humidity
      - name: "Outdoor Humidity (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.humidity }}"
        unit_of_measurement: "%"
        state_class: "measurement"
        device_class: "humidity"
        unique_id: "sensor.wind-station.humidity"
    # Pressure
      - name: "Outdoor Pressure (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.pressure }}"
        unit_of_measurement: "mbar"
        state_class: "measurement"
        device_class: "pressure"
        unique_id: "sensor.wind-station.pressure"
    # Voltage
      - name: "Battery voltage (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.voltage }}"
        unit_of_measurement: "V"
        state_class: "measurement"
        device_class: "voltage"
        unique_id: "sensor.wind-station.voltage"
    # Rain
      - name: "Rain (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.rain }}"
        unit_of_measurement: "mm"
        state_class: "measurement"
        device_class: "precipitation"
        unique_id: "sensor.wind-station.rain"
    # Wind Speed
      - name: "Wind Speed (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.wind_speed }}"
        unit_of_measurement: "m/s"
        state_class: "measurement"
        device_class: "wind_speed"
        unique_id: "sensor.wind-station.wind_speed"
    # Wind Direcction
      - name: "Wind Direction (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.wind_direction }}"
        unit_of_measurement: "°"
        state_class: "measurement"
        unique_id: "sensor.wind-station.wind_direction"
    # Light
      - name: "Outdoor Light Level (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.light }}"
        unit_of_measurement: "lx"
        state_class: "measurement"
        device_class: "illuminance"
        unique_id: "sensor.wind-station.light"
    # Rain Per Second
      - name: "Rain Per Second (wind-station)"
        state_topic: "enviro/wind-station"
        value_template: "{{ value_json.readings.rain_per_second }}"
        unit_of_measurement: "mm"
        state_class: "measurement"
        device_class: "precipitation"
        unique_id: "sensor.wind-station.rain_per_second"`

andymanic, 2023-03-05:
#156 Just popping back to say I did it :)