pimoroni/enviro #162: Enviro weather hangs after a few hours

thisrandomuser, 2023-03-24:
After setting up an enviro weather board with the 0.0.9 firmware, powered over USB, I repeated had problems with it stopping sending MQTT message after a few hours. The exact time seemed to vary each time, but it typically seemed to be between 4-7 hours. When I went to check it, the red warning LED was lit. 

The system is mounted outside with winf and rain modules attached, but is within 3m of the nearest router, which is just inside a workshop roof. i have proviously used other wifi boards for weather readings in this configuration that have worked fine. I also still have other wifi monitors (not pi picos) outside that transmit and operate reliably. 

So far I have tried the folllowing things:-
Writing a simplified micropython code that provides the basic weather reading and MQTT transmisions over wifi. While I got the code to work, I still ended up with similar issues of the system stopping transmitting after a short period. By adding a watchdog timeer, I could get the board to reset, which gave me a couple of readings ever 15-20 minutes, but still wasn't ideal.

Having read that some pico Ws have had issues with wifi connectivity, and have a habit of crashing when they lose their connections, I have laso treid a different approach. I soldered connections for a basic 433 radio onto GPIO16 and the power and gnd pins, went back to Arduino and wrote another version of my weather software to transmit over the 433mhz radio using the Radiohead library. I then pick up the signals using a SDR radio module and rtl_433 library on a Pi4 board. This has now worked reliably for a week sending readings every minute, including additional data from lux and uv sensors wired through the QUIC port.

From these tests, I believe that the connection problems are not enviro board related, or necessarily due to a problem in the 0.0.9 software. 

Are you aware of issues with reliability of the wifi on the pi pico Ws boards? Is there anything that you could add into the standard enviro weather software to try to mitigate these issues and help maintain the connection? 

I'm sure I have seen similar issues raised elsewhere, but I cannot find them on here at the moment. Please move this post if there is already an appropriate open issue to link it to.

jfarcher, 2023-04-26:
I'm also seeing this. After a few hours it just stops. I've noticed that the white LED comes on solid, and battery drains quick too.

thisrandomuser, 2023-04-26:
I was running off a usb adapter. My arduino IDE code with the 433Mhz radio has been running without my needing to intervene since my original post in March though. The only thing I'm missing is rainfall, but that seems to be a coding error of mine in the way I read my interrupt counter.

basil-dsouza, 2024-01-06:
I've been facing this same problem for quite a while as well. Have had version 0.0.8 all the way through the latest version (0.0.10) having the same issues. I am using an enviro indoor, so i think this is common to the entire family of devices.
Is a bit frustrating to have to keep coming back to a fully drained device and keep changing batteries every few days and miss out on readings.

sjefferson99, 2024-01-07:
I experienced the solid white light lock up after a few hours when polling frequently issue, I explored my network and stability issues in #119 and #87. This was resolved for me by putting in improvements to the wifi connection code created by mrexodia, which I have captured in an as yet not merged PR #199.

I recommend giving that code a go for any network or general instability issues.