pimoroni/enviro #167: running two nodes = duplicate hostname/dns name on DHCP server

ikonia, 2023-03-28:
I've got 2 enviro local on the same wlan, the dhcp see's them both ask for addresses and updates dns successfully, however both of them have the same default hostname that doesn't seem to be able to be changed
eg:
DHCPDISCOVER from 28:c1:05:02:9d (PYBD) via enp2s0
DHCPDISCOVER from 28:c1:05:22:82 (PYBD) via enp2s0

is there anyway to get them to identify themselves differently.

I'm also not sure if this is causing any problems in terms of wifi stability either due to conflicting FQDN as dhcp will also update DDNS with FQDN from dhcp requests.

ikonia, 2024-04-23:
this is confirmed causing a problem, as DDNS is unable to handle duplicate hostnames and as such will refuse to offer an IP address if the forward map cannot be set.

Gadgetoid, 2024-04-23:
Which part of the above is the hostname?

Looks like a fix is possible downstream- https://github.com/micropython/micropython/pull/10635

I guess we could change the hostname to include part of the Pico's UUID. Something like (note: :warning: test this in a REPL first I don't know what format UUID will be in!):

```
UUID = machine.unique_id()
machine.hostname(f"enviro{UUID[-4:]}")
```

You might be able to make this change early in your board's `main.py` and see if it accomplishes what we'd want it to.

ikonia, 2024-04-23:
the hostname being advertised by the network interface is "PYBD" so in essence 2 devices are called PYBD

sjefferson99, 2024-04-23:
@Gadgetoid As my board is misbehaving on my custom branch code, I've flashed it with this UF2 to properly test the new firmware and wifi updates. I can confirm that now I've taken my MAC reservation off the DHCP server I immediately get a hostname of EnviroW-7822.

Assuming you have picked the end of the UUID that is likely to change most (I have no idea how it's allocated to the flash chip), then this looks to be successful. Providing no one else finds they have the same suffix, this should be enough for the use case.

Gadgetoid, 2024-04-24:
> Assuming you have picked the end of the UUID that is likely to change most

I don't know for sure, but I think it's convention to have the bit that changes most on the right.

I think this is safer, and easier to implement without extensive testing, than taking user input as the hostname. Though the latter is still possible.

commit 33ab1a20 (2024-04-23): Make hostname unique per board.

(Attempt to) make the hostname unique per board, by using the last
four chars of the unique ID (flash chip ID) as a suffix.

Fixes #167
PR #218 (2024-04-24): Make hostname unique per board.
(Attempt to) make the hostname unique per board, by using the last four chars of the unique ID (flash chip ID) as a suffix.

Fixes #167