pimoroni/enviro #168: Would like to have rain per hour and rain today values

sjefferson99, 2023-03-29:
Weather underground only takes per hour and per day values for rain and these are useful values in their own right.

ZodiusInfuser, 2023-03-31:
Hi @sjefferson99. These sound pretty easy to add, if all that's needed is scaling up the current measurement (so the rain per hour becomes an estimate). Recording the cumulative rain over a full hour our day would be a bit more complicated though.

sjefferson99, 2023-03-31:
@ZodiusInfuser Apologies, I wasn't sure of the contribution approach so I created an issue for any PRs that didn't already have one to denote the reasoning behind them and tagged the issue in the PR. I'll make a clearer comment next time.