pimoroni/enviro #170: Enviro Urban Noise Sensor

Astro-Pete, 2023-03-30:
I seem to get a constant 1.5V reading. Is this sensor functional? Even if i shout at it while it takes a reading it only jumps to say 1.53V

Also is there any info on how to convert this reading to dB?

Astro-Pete, 2023-03-30:
OK, after some more digging around I found the #98 thread with info on the bug in urban.py about the incorrect / 3.3 * 65535 line. Would still like to know how V relates to dB though??

I’ve also downloaded the newer firmware as I had the failed to connect to wifi issue too.

I didn’t see a thread specific to the noise sensor issue so that’s why I posted.

ZodiusInfuser, 2023-03-31:
Hi @Astro-Pete. As you spotted, there was an error in the code, that was subsequently fixed in patch v0.0.9.

As for relating V to dB, that's something I looked into but did not find a definitive answer for how to do that conversion (since V is an absolute measurement, and dB is a relative measurement). I would happily accept a pull request from someone knowledgeable in the area.

soficshifts, 2024-02-23:
Will nudge this as I know there were issues with GitHub and my comments appearing when I posted the above.

The formula (and its derivation) is explained very long-windedly here:

https://github.com/soficshifts/pirmoroni/blob/main/documentation/enviro-urban/enviro-urban-volts-to-decibels.md

This could be put in the enviro-urban code, but I object to doing so on the premise that a Scientific Instrument should return the raw data, what you do with that post-processing is your business.  

Secondly, the formula is an approximation as the Specification sheet states the gain is determined by the ratio of two resistors that the installer would have provided.  So the `+35` offset was taken by calibrating against a smartphone app, if these resistors are known, one could get a more accurate formula.  

So someone is welcome to change the code, as I probably am not motivated to do so myself for the above reasons. I'd suggest a config parameter that allows you to choose decibels or the raw voltage reading.

@ZodiusInfuser  - I'm happy to make a more succinct writeup for the doco as a pull request.

Thanks, I hope this helps.