pimoroni/enviro #175: Readings filling up filesystem

doug-burrell, 2023-04-29:
I've got an Enivro Indoor that I've been running since early February of 2023 and I'm finding that the readings are filling up the Pico's filesystem. Once it gets to the low storage limit it stops working until I manually remove all the readings files.

MrDrem, 2023-04-29:
Do you have it set to upload to somewhere? If not this is expected behaviour I believe. It might be better if it overwrote oldest readings maybe.

doug-burrell, 2023-05-01:
It's uploading to Adafruit successfully. I can see in main.py it's saving the readings but eventually it fills up because there's no file rotation for the readings. I've commented out this line of code for now but there should be some sort of file rotation going on to prevent the readings folder from filling up the filesystem.

sjefferson99, 2023-05-04:
File deletion is handled in upload_readings() in enviro/__init__.py on line 428 on the main branch. This calls function calls the appropriate destination code, so it should work the same for all valid destinations and it certainly works for my influxdb and wunderground configs.

It is designed to create log files only as a buffer and only store them as long as is needed until an upload is triggered AND successful. As a deliberate side effect, if you don't specify a destination, it will log locally, but I don't think there's any rotation for that scenario as yet, not a mode I have worked with though.

I would guess that you are not getting a 100% successful upload for it to skip this line, which should be evident in the log file where you see a log.info line in that segment that isn't " - uploaded <logfilename>". If you can find the section of upload_readings() your code is hitting, you can hopefully correct the upload issue or fix the bug in that scenario, as you say it is uploading data to adafruit OK.

doug-burrell, 2023-05-08:
It looks like the problem has been fixed in the main branch, I'm using v0.0.2 which is the default release firmware that came on my unit and it does both upload and save readings locally, whereas main only saves readings locally if there's no upload configuration. I'll have to upgrade mine sometime soon. Thanks for the help!