pimoroni/enviro #176: Fine Offset windspeed sensor may produce 4 ticks per revolution, not 2 as you assume.

learnfromgirls, 2023-05-18:
Fine Offset windspeed sensor may produce 4 ticks per revolution, not 2 as you assume.
It has a cunning magnet placement that operates reed switch twice per rev as discussed in.
https://cumulus.hosiene.co.uk/viewtopic.php?p=35990#p35990
Gina says:- I have now carried out the definitive test. The answer is TWO pulses per revolution. I was wrong. :oops: I hadn't actually done the test carefully enough - I had assumed 1 magnet and 1 reed switch meant 1 pulse per rev.

I think your tick means a state change, i.e low to high or high to low,  So one pulse is 2 of your state changes. 2 pulses will be 4 of your state changes.
Your WIND_FACTOR also seems optimistic if it is intended to be the anemometer factor, it should be closer to 0.03. Though your low WIND_FACTOR currently compensates for the factor of 2 error in ticks per rotation as it effectively is 0.0436 (probably too high).
Ideally the anemometer factor should be measured by someone with a known wind.

# scaling factor for wind speed in m/s
WIND_FACTOR = 0.0218

Your code in weather.py says:-
# work out rotation speed in hz (two ticks per rotation)
  rotation_hz = (1000 / average_tick_ms) / 2

MrDrem, 2024-04-26:
I've been looking at the code for the wind speed, looking to improve it, and to try and work out why it makes my board fall over some times, and I may have found something.

I switched it so that it was only logging the ticks, and found the following:

```
2024-04-26 22:21:27 [exception / 126kB] > wind speed stats - ticks: [3126]
2024-04-26 22:22:04 [exception / 126kB] > wind speed stats - ticks: [2906, 2906, 2906]
2024-04-26 22:22:04 [exception / 122kB] ! Traceback (most recent call last):
  File "main.py", line 71, in <module>
  File "enviro/__init__.py", line 411, in get_sensor_readings
  File "enviro/boards/weather.py", line 194, in get_sensor_readings
  File "enviro/boards/weather.py", line 119, in wind_speed
ZeroDivisionError: divide by zero

2024-04-26 22:24:04 [exception / 126kB] > wind speed stats - ticks: [2706, 3119, 3296, 3296, 3297, 3675]
2024-04-26 22:26:04 [exception / 126kB] > wind speed stats - ticks: [3290]
2024-04-26 22:28:04 [exception / 126kB] > wind speed stats - ticks: [3347]
2024-04-26 22:30:04 [exception / 126kB] > wind speed stats - ticks: [2592, 2780, 3186, 3411, 3411, 3411]
2024-04-26 22:32:04 [exception / 126kB] > wind speed stats - ticks: [3008, 3008, 3008]
2024-04-26 22:32:04 [exception / 122kB] ! Traceback (most recent call last):
  File "main.py", line 71, in <module>
  File "enviro/__init__.py", line 411, in get_sensor_readings
  File "enviro/boards/weather.py", line 194, in get_sensor_readings
  File "enviro/boards/weather.py", line 119, in wind_speed
ZeroDivisionError: divide by zero

2024-04-26 22:34:04 [exception / 126kB] > wind speed stats - ticks: [2891, 3076, 3076, 3076, 3476]
2024-04-26 22:36:04 [exception / 126kB] > wind speed stats - ticks: [2906, 2906, 2906]
2024-04-26 22:36:04 [exception / 122kB] ! Traceback (most recent call last):
  File "main.py", line 71, in <module>
  File "enviro/__init__.py", line 411, in get_sensor_readings
  File "enviro/boards/weather.py", line 194, in get_sensor_readings
  File "enviro/boards/weather.py", line 119, in wind_speed
ZeroDivisionError: divide by zero

2024-04-26 22:38:04 [exception / 126kB] > wind speed stats - ticks: [3017, 3017, 3017]
2024-04-26 22:38:04 [exception / 122kB] ! Traceback (most recent call last):
  File "main.py", line 71, in <module>
  File "enviro/__init__.py", line 411, in get_sensor_readings
  File "enviro/boards/weather.py", line 194, in get_sensor_readings
  File "enviro/boards/weather.py", line 119, in wind_speed
ZeroDivisionError: divide by zero

2024-04-26 22:40:04 [exception / 126kB] > wind speed stats - ticks: [2788, 3129]
2024-04-26 22:42:04 [exception / 126kB] > wind speed stats - ticks: [2750, 3138]
2024-04-26 22:44:04 [exception / 126kB] > wind speed stats - ticks: [2810, 3025, 3025, 3025, 3469]
2024-04-26 22:46:04 [exception / 126kB] > wind speed stats - ticks: [3088]
```

Sometimes, I suspect when the wind is very low, the anemometer seems to sit and flick the reed switch really fast.

I'm rewriting my code so that if ticks(-1) == ticks(0) it will return a wind speed of zero, but this will have implications for how wind speed is measured overall.

As above, I'm now wondering if the code that we have counts the switch opening as one change, and it closing as another? If this is the case, can we tell the difference? Is this something that @Gadgetoid or @sjefferson99 know?

MrDrem, 2024-04-27:
So, looking at the above, I've realised that this line:

`2024-04-26 22:44:04 [exception / 126kB] > wind speed stats - ticks: [2810, 3025, 3025, 3025, 3469]`

would also cause the average wind speed to be incorrect. This would give (3469-2810)/5, when it should give it (3469-2810)/3.

I've rewritten my code to remove the duplicates, and it's still working. I'm currently certain that my code isn't giving the correct speeds, so I'm not ready to share it in public yet.

gbravery, 2024-10-01:
I'm interested in how you got on with trying to debug wind speed calculations?
I have a situation where my ticks where: [12711, 12711, 12712]
This resulted in a speed of 958.8m/s - Which is obviously nonsense!

Digging about, I have found that the max speed recorded was 253mph -> 113.101m/s
Using that I've derived that the average_tick_ms would need to be: 4.238752...

I am therefore going to make my code look for an average_tick_ms of above 4

I'm also wondering if the tick while statement should ignore ticks where tick is the same time...?
In my case, I think the while collection loop would have then timed out, and I'd have gotten a correct "zero" speed instead...